RainerKuemmerle/g2o #841: Segmentaion fault for large scale dataset

uestcyile, 2024-11-04:
I tried to solve a slam problem by G2O, and it seems work well on small dataset. But when I tried to solve a large scale problem, the  segmetation faut occurred, and the debug info by gdb is as follows. Can you give me some debugging advices, thanks.

Thread 1 "QMapServerNslEx" received signal SIGSEGV, Segmentation fault.
0x00007ffff7819038 in Eigen::internal::minimum_degree_ordering<double, int> (C=..., perm=...) at /usr/include/eigen3/Eigen/src/OrderingMethods/Amd.h:353
353             for(p = Cp[i]+1; p <= Cp[i] + ln-1; p++) w[Ci[p]] = mark;
(gdb) bt
#0  0x00007ffff7819038 in Eigen::internal::minimum_degree_ordering<double, int> (C=..., perm=...) at /usr/include/eigen3/Eigen/src/OrderingMethods/Amd.h:353
#1  0x00007ffff7829868 in Eigen::AMDOrdering<int>::operator()<Eigen::SparseMatrix<double, 0, int> > (this=<synthetic pointer>, perm=..., mat=...)
    at /usr/include/x86_64-linux-gnu/bits/string_fortified.h:71
#2  Eigen::SimplicialCholeskyBase<Eigen::SimplicialLDLT<Eigen::SparseMatrix<double, 0, int>, 2, Eigen::AMDOrdering<int> > >::ordering (this=this@entry=0x7fc8ecf96a98, a=...,
    pmat=@0x7fffffffd930: 0x7fffffffd980, ap=...) at /usr/include/eigen3/Eigen/src/SparseCholesky/SimplicialCholesky.h:671
#3  0x00007ffff782a187 in Eigen::SimplicialCholeskyBase<Eigen::SimplicialLDLT<Eigen::SparseMatrix<double, 0, int>, 2, Eigen::AMDOrdering<int> > >::analyzePattern (doLDLT=true, a=...,
    this=0x7fc8ecf96a98) at /usr/include/x86_64-linux-gnu/bits/string_fortified.h:71
#4  Eigen::SimplicialLDLT<Eigen::SparseMatrix<double, 0, int>, 2, Eigen::AMDOrdering<int> >::analyzePattern (a=..., this=0x7fc8ecf96a98)
    at /usr/include/eigen3/Eigen/src/SparseCholesky/SimplicialCholesky.h:483
#5  g2o::LinearSolverEigen<Eigen::Matrix<double, -1, -1, 0, -1, -1> >::computeSymbolicDecomposition (this=this@entry=0x7fc8ecf96a40, A=...)
    at /zhyile/code/nsl/qmapserver/nsl/libs/ThirdParty/g2o/solvers/linear_solver_eigen.h:149
#6  0x00007ffff782a900 in g2o::LinearSolverEigen<Eigen::Matrix<double, -1, -1, 0, -1, -1> >::solve (this=0x7fc8ecf96a40, A=..., x=0x7ff97bb6be60, b=0x7ff7740eca60)
    at /usr/include/c++/9/ext/new_allocator.h:89
#7  0x00007ffff78356b4 in g2o::BlockSolver<g2o::BlockSolverTraits<-1, -1> >::solve (this=0x7fffe8cb11c0) at /usr/include/c++/9/bits/unique_ptr.h:619
#8  0x00007ffff336329e in g2o::OptimizationAlgorithmLevenberg::solve (this=0x7fef704af100, iteration=<optimized out>, online=<optimized out>)
    at /zhyile/code/nsl/qmapserver/nsl/libs/ThirdParty/g2o/core/optimization_algorithm_levenberg.cpp:107
#9  0x00007ffff33679f3 in g2o::SparseOptimizer::optimize (this=this@entry=0x7fffffffde10, iterations=iterations@entry=20, online=online@entry=false)
    at /zhyile/code/nsl/qmapserver/nsl/libs/ThirdParty/g2o/core/sparse_optimizer.cpp:400

RainerKuemmerle, 2024-11-17:
For large scale problems I would recommend Cholmod as solver.
The above stacktrace is inside Eigen's AMD ordering.
Can you provide data or more details on how to reproduce the problem?

uestcyile, 2024-11-20:
Sorry, I cannot provide data for information security reasons. But I can provide more details.
Firstly, my optimizer has about 650000 vertices and 3400000 edges in total, the data size is a bit large.
As you said, I tried replace linear solver with Cholmod and CSparse. And both of them can finish optimization without segmentation fault. But  in terms of actual optimization results on my data, CSparse is better than Cholmod for its convergence speed and performance.
Thanks for your advice.

uestcyile, 2024-11-20:
By the way, is the Eigen solver suitable for small and medium-sized problems? Cholmod and CSparse solver are suitable for large scale problem, right?

closed 2024-11-20 by uestcyile