RainerKuemmerle/g2o #844: NAN error in g2o::optimizer::optimize(), anyone could help me?

guoxxiong, 2024-11-22:
```
bandpt position: 0.975, -0.364369
measurement: 0.975, -0.364369
bandpt position: 0.975, -0.465039
measurement: 0.975, -0.465039
bandpt position: 0.975, -0.566058
measurement: 0.975, -0.566058
bandpt position: 0.975, -0.667078
measurement: 0.975, -0.667078
bandpt position: 0.975, -0.869117
measurement: 0.975, -0.768098
bandpt position: 0.975, -0.869117
measurement: 0.975, -0.869117
bandpt position: 0.975, -0.970137
measurement: 0.975, -0.970137
bandpt position: 0.975, -0.970137
measurement: 0.975, -1.07116
bandpt position: 0.975, -1.17218
measurement: 0.975, -1.17218
bandpt position: 0.975, -1.2732
measurement: 0.975, -1.2732
bandpt position: 0.975, -1.37422
measurement: 0.975, -1.37422
bandpt position: 0.975, -1.37422
measurement: 0.975, -1.47511
bandpt position: 0.975, -1.58822
measurement: 0.975, -1.58822
bandpt position: 0.975, -1.58822
measurement: 0.975, -1.69323
bandpt position: 0.975, -0.364369
measurement: 0.975, -0.364369
bandpt position: 0.975, -0.364369
measurement: 0.975, -0.364369
bandpt position: 0.975, -0.364369
measurement: 0.975, -0.364369
bandpt position: 0.975, -0.364369
measurement: 0.975, -0.364369
bandpt position: 0.975, -0.364369
measurement: 0.975, -0.364369
bandpt position: 0.975, -0.364369
measurement: 0.975, -0.364369
bandpt position: 0.975, -0.465039
measurement: 0.975, -0.465039
bandpt position: 0.975, -0.465039
measurement: 0.975, -0.465039
bandpt position: 0.975, -0.465039
measurement: 0.975, -0.465039
bandpt position: 0.975, -0.465039
measurement: 0.975, -0.465039
bandpt position: 0.975, -0.465039
measurement: 0.975, -0.465039
bandpt position: 0.975, -0.465039
measurement: 0.975, -0.465039
bandpt position: 0.975, -0.566058
measurement: 0.975, -0.566058
bandpt position: 0.975, -0.566058
measurement: 0.975, -0.566058
bandpt position: 0.975, -0.566058
measurement: 0.975, -0.566058
bandpt position: 0.975, -0.566058
measurement: 0.975, -0.566058
bandpt position: 0.975, -0.566058
measurement: 0.975, -0.566058
bandpt position: 0.975, -0.566058
measurement: 0.975, -0.566058
bandpt position: 0.975, -0.667078
measurement: 0.975, -0.667078
bandpt position: 0.975, -0.667078
measurement: 0.975, -0.667078
bandpt position: 0.975, -0.667078
measurement: 0.975, -0.667078
bandpt position: 0.975, -0.667078
measurement: 0.975, -0.667078
bandpt position: 0.975, -0.667078
measurement: 0.975, -0.667078
bandpt position: 0.975, -0.667078
measurement: 0.975, -0.667078
bandpt position: 0.975, -0.869117
measurement: 0.975, -0.768098
bandpt position: 0.975, -0.869117
measurement: 0.975, -0.768098
bandpt position: 0.975, -0.869117
measurement: 0.975, -0.768098
bandpt position: 0.975, -0.869117
measurement: 0.975, -0.768098
bandpt position: 0.975, -0.869117
measurement: 0.975, -0.768098
bandpt position: 0.975, -0.869117
measurement: 0.975, -0.768098
bandpt position: 0.975, -0.869117
measurement: 0.975, -0.869117
bandpt position: 0.975, -0.869117
measurement: 0.975, -0.869117
bandpt position: 0.975, -0.869117
measurement: 0.975, -0.869117
bandpt position: 0.975, -0.869117
measurement: 0.975, -0.869117
bandpt position: 0.975, -0.869117
measurement: 0.975, -0.869117
bandpt position: 0.975, -0.869117
measurement: 0.975, -0.869117
bandpt position: 0.975, -0.970137
measurement: 0.975, -0.970137
bandpt position: 0.975, -0.970137
measurement: 0.975, -0.970137
bandpt position: 0.975, -0.970137
measurement: 0.975, -0.970137
bandpt position: 0.975, -0.970137
measurement: 0.975, -0.970137
bandpt position: 0.975, -0.970137
measurement: 0.975, -0.970137
bandpt position: 0.975, -0.970137
measurement: 0.975, -0.970137
bandpt position: 0.975, -0.970137
measurement: 0.975, -1.07116
bandpt position: 0.975, -0.970137
measurement: 0.975, -1.07116
bandpt position: 0.975, -0.970137
measurement: 0.975, -1.07116
bandpt position: 0.975, -0.970137
measurement: 0.975, -1.07116
bandpt position: 0.975, -0.970137
measurement: 0.975, -1.07116
bandpt position: 0.975, -0.970137
measurement: 0.975, -1.07116
bandpt position: 0.975, -1.17218
measurement: 0.975, -1.17218
bandpt position: 0.975, -1.17218
measurement: 0.975, -1.17218
bandpt position: 0.975, -1.17218
measurement: 0.975, -1.17218
bandpt position: 0.975, -1.17218
measurement: 0.975, -1.17218
bandpt position: 0.975, -1.17218
measurement: 0.975, -1.17218
bandpt position: 0.975, -1.17218
measurement: 0.975, -1.17218
bandpt position: 0.975, -1.2732
measurement: 0.975, -1.2732
bandpt position: 0.975, -1.2732
measurement: 0.975, -1.2732
bandpt position: 0.975, -1.2732
measurement: 0.975, -1.2732
bandpt position: 0.975, -1.2732
measurement: 0.975, -1.2732
bandpt position: 0.975, -1.2732
measurement: 0.975, -1.2732
bandpt position: 0.975, -1.2732
measurement: 0.975, -1.2732
bandpt position: 0.975, -1.37422
measurement: 0.975, -1.37422
bandpt position: 0.975, -1.37422
measurement: 0.975, -1.37422
bandpt position: 0.975, -1.37422
measurement: 0.975, -1.37422
bandpt position: 0.975, -1.37422
measurement: 0.975, -1.37422
bandpt position: 0.975, -1.37422
measurement: 0.975, -1.37422
bandpt position: 0.975, -1.37422
measurement: 0.975, -1.37422
bandpt position: 0.975, -1.37422
measurement: 0.975, -1.47511
bandpt position: 0.975, -1.37422
measurement: 0.975, -1.47511
bandpt position: 0.975, -1.37422
measurement: 0.975, -1.47511
bandpt position: 0.975, -1.37422
measurement: 0.975, -1.47511
bandpt position: 0.975, -1.37422
measurement: 0.975, -1.47511
bandpt position: 0.975, -1.37422
measurement: 0.975, -1.47511
bandpt position: 0.975, -1.58822
measurement: 0.975, -1.58822
bandpt position: 0.975, -1.58822
measurement: 0.975, -1.58822
bandpt position: 0.975, -1.58822
measurement: 0.975, -1.58822
bandpt position: 0.975, -1.58822
measurement: 0.975, -1.58822
bandpt position: 0.975, -1.58822
measurement: 0.975, -1.58822
bandpt position: 0.975, -1.58822
measurement: 0.975, -1.58822
bandpt position: 0.975, -1.58822
measurement: 0.975, -1.69323
bandpt position: 0.975, -1.58822
measurement: 0.975, -1.69323
bandpt position: 0.975, -1.58822
measurement: 0.975, -1.69323
bandpt position: 0.975, -1.58822
measurement: 0.975, -1.69323
bandpt position: 0.975, -1.58822
measurement: 0.975, -1.69323
bandpt position: 0.975, -1.58822
measurement: 0.975, -1.69323
bandpt position: -nan, -nan
measurement: 0.975, -0.364369
teb_local_planner/g2o_types/edge_via_point.h
	line = 90
	cond = std::isfinite(_error[0])
	message = 
[FATAL] [1732258518.080993652]: EdgeViaPoint::computeError() _error[0]=-nan
```

RainerKuemmerle, 2024-11-23:
I think, you are using the TEB planner inside ROS. You need to check the implementation there.
https://github.com/rst-tu-dortmund/teb_local_planner/blob/noetic-devel/include/teb_local_planner/g2o_types/edge_via_point.h#L88

Maybe no measurement is set, a wrong measurement, the band generated a NaN beforehand.

closed 2024-11-23 by RainerKuemmerle

guoxxiong, 2024-11-25:
Hi, I found that an inf is included with "_b" and "_hessian" in vertices, but the last "_estimate" before this is normal. I 'm guessing what causes this phenomenon in general. Would you like to give some suggestions?

guoxxiong, 2024-11-25:
@RainerKuemmerle 
Hi,
I found that an inf is included with "_b" and "_hessian" in vertices, but the last "_estimate" before this is normal. I 'm guessing what causes this phenomenon in general. Would you like to give some suggestions? I have checked the input data was correct.