RainerKuemmerle/g2o #847: Memory leak?

QiAnXinCodeSafe, 2024-11-25:
Hi all,
This is Qianxin CodeSafe Team, we found a suspicious issue, allocating memory at

https://github.com/RainerKuemmerle/g2o/blob/9aebe10343e4e3605eb0ba7b2eef6aa8d53695a6/g2o/examples/ba_anchored_inverse_depth/ba_anchored_inverse_depth_demo.cpp#L201 

but , Assume the condition value is false,which means the loop body executes 0 times,at

https://github.com/RainerKuemmerle/g2o/blob/9aebe10343e4e3605eb0ba7b2eef6aa8d53695a6/g2o/examples/ba_anchored_inverse_depth/ba_anchored_inverse_depth_demo.cpp#L203

this will result in a memory leak of point “v_p”

RainerKuemmerle, 2025-01-01:
Seems like you are right here. As this is "just" an example, I will probably not fix this issue. PR is welcome though.
If you find other / similar issues in the code base, feel free to create another issue. Thank you.

closed 2025-03-17 by github-actions[bot]

wangzheqie, 2025-03-17:
这是来自QQ邮箱的自动回复邮件。你好，我已经收到你的邮件，尽快处理。