RainerKuemmerle/g2o #857: Compilation Error undefined reference to `g2o::OptimizationAlgorithmG

eduardoavila1m, 2024-12-26:
Trying to compile some Visual Od Code.

The API is complaining about not finding the .h files. However I've link them.

C:/msys64/ucrt64/bin/../lib/gcc/x86_64-w64-mingw32/13.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: CMakeFiles\g2oCurveFitting.dir/objects.a(g2oCurveFitting.cpp.obj):g2oCurveFitting.cpp:(.text+0x227): undefined reference to `g2o::OptimizationAlgorithmGaussNewton::OptimizationAlgorithmGaussNewton(std::unique_ptr<g2o::Solver, std::default_delete<g2o::Solver> >)'
C:/msys64/ucrt64/bin/../lib/gcc/x86_64-w64-mingw32/13.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: CMakeFiles\g2oCurveFitting.dir/objects.a(g2oCurveFitting.cpp.obj):g2oCurveFitting.cpp:(.text+0x267): undefined reference to `g2o::SparseOptimizer::SparseOptimizer()'
C:/msys64/ucrt64/bin/../lib/gcc/x86_64-w64-mingw32/13.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: CMakeFiles\g2oCurveFitting.dir/objects.a(g2oCurveFitting.cpp.obj):g2oCurveFitting.cpp:(.text+0x27a): undefined reference to `g2o::SparseOptimizer::setAlgorithm(g2o::OptimizationAlgorithm*)'
C:/msys64/ucrt64/bin/../lib/gcc/x86_64-w64-mingw32/13.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: CMakeFiles\g2oCurveFitting.dir/objects.a(g2oCurveFitting.cpp.obj):g2oCurveFitting.cpp:(.text+0x28b): undefined reference to `g2o::SparseOptimizer::setVerbose(bool)'
C:/msys64/ucrt64/bin/../lib/gcc/x86_64-w64-mingw32/13.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: CMakeFiles\g2oCurveFitting.dir/objects.a(g2oCurveFitting.cpp.obj):g2oCurveFitting.cpp:(.text+0x3a1): undefined reference to `g2o::HyperGraph::Edge::setId(int)'
C:/msys64/ucrt64/bin/../lib/gcc/x86_64-w64-mingw32/13.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: CMakeFiles\g2oCurveFitting.dir/objects.a(g2oCurveFitting.cpp.obj):g2oCurveFitting.cpp:(.text+0x4a8): undefined reference to `g2o::OptimizableGraph::addEdge(g2o::OptimizableGraph::Edge*)'
C:/msys64/ucrt64/bin/../lib/gcc/x86_64-w64-mingw32/13.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: CMakeFiles\g2oCurveFitting.dir/objects.a(g2oCurveFitting.cpp.obj):g2oCurveFitting.cpp:(.text+0x506): undefined reference to `g2o::SparseOptimizer::initializeOptimization(int)'
C:/msys64/ucrt64/bin/../lib/gcc/x86_64-w64-mingw32/13.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: CMakeFiles\g2oCurveFitting.dir/objects.a(g2oCurveFitting.cpp.obj):g2oCurveFitting.cpp:(.text+0x51d): undefined reference to `g2o::SparseOptimizer::optimize(int, bool)'
C:/msys64/ucrt64/bin/../lib/gcc/x86_64-w64-mingw32/13.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: CMakeFiles\g2oCurveFitting.dir/objects.a(g2oCurveFitting.cpp.obj):g2oCurveFitting.cpp:(.text+0x636): undefined reference to `g2o::SparseOptimizer::~SparseOptimizer()'
C:/msys64/ucrt64/bin/../lib/gcc/x86_64-w64-mingw32/13.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: CMakeFiles\g2oCurveFitting.dir/objects.a(g2oCurveFitting.cpp.obj):g2oCurveFitting.cpp:(.text+0x6ef): undefined reference to `g2o::SparseOptimizer::~SparseOptimizer()'
C:/msys64/ucrt64/bin/../lib/gcc/x86_64-w64-mingw32/13.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: CMakeFiles\g2oCurveFitting.dir/objects.a(g2oCurveFitting.cpp.obj):g2oCurveFitting.cpp:(.text$_ZN3g2o16OptimizableGraph9addVertexEPNS0_6VertexE[_ZN3g2o16OptimizableGraph9addVertexEPNS0_6VertexE]+0x22): undefined reference to `g2o::OptimizableGraph::addVertex(g2o::OptimizableGraph::Vertex*, g2o::HyperGraph::Data*)'
C:/msys64/ucrt64/bin/../lib/gcc/x86_64-w64-mingw32/13.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: CMakeFiles\g2oCurveFitting.dir/objects.a(g2oCurveFitting.cpp.obj):g2oCurveFitting.cpp:(.text$_ZN3g2o15BlockSolverBaseD2Ev[_ZN3g2o15BlockSolverBaseD2Ev]+0x22): undefined reference to `g2o::Solver::~Solver()'
C:/msys64/ucrt64/bin/../lib/gcc/x86_64-w64-mingw32/13.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: CMakeFiles\g2oCurveFitting.dir/objects.a(g2oCurveFitting.cpp.obj):g2oCurveFitting.cpp:(.text$_ZN3g2o10BaseVertexILi3EN5Eigen6MatrixIdLi3ELi1ELi0ELi3ELi1EEEED2Ev[_ZN3g2o10BaseVertexILi3EN5Eigen6MatrixIdLi3ELi1ELi0ELi3ELi1EEEED2Ev]+0x43): undefined reference to `g2o::OptimizableGraph::Vertex::~Vertex()'
C:/msys64/ucrt64/bin/../lib/gcc/x86_64-w64-mingw32/13.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: CMakeFiles\g2oCurveFitting.dir/objects.a(g2oCurveFitting.cpp.obj):g2oCurveFitting.cpp:(.text$_ZN3g2o8BaseEdgeILi1EdED2Ev[_ZN3g2o8BaseEdgeILi1EdED2Ev]+0x31): undefined reference to `g2o::OptimizableGraph::Edge::~Edge()'
C:/msys64/ucrt64/bin/../lib/gcc/x86_64-w64-mingw32/13.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: CMakeFiles\g2oCurveFitting.dir/objects.a(g2oCurveFitting.cpp.obj):g2oCurveFitting.cpp:(.text$_ZN3g2o10BaseVertexILi3EN5Eigen6MatrixIdLi3ELi1ELi0ELi3ELi1EEEEC2Ev[_ZN3g2o10BaseVertexILi3EN5Eigen6MatrixIdLi3ELi1ELi0ELi3ELi1EEEEC2Ev]+0x17): undefined reference to `g2o::OptimizableGraph::Vertex::Vertex()'
C:/msys64/ucrt64/bin/../lib/gcc/x86_64-w64-mingw32/13.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: CMakeFiles\g2oCurveFitting.dir/objects.a(g2oCurveFitting.cpp.obj):g2oCurveFitting.cpp:(.text$_ZN3g2o10BaseVertexILi3EN5Eigen6MatrixIdLi3ELi1ELi0ELi3ELi1EEEEC2Ev[_ZN3g2o10BaseVertexILi3EN5Eigen6MatrixIdLi3ELi1ELi0ELi3ELi1EEEEC2Ev]+0xbf): undefined reference to `g2o::OptimizableGraph::Vertex::~Vertex()'
C:/msys64/ucrt64/bin/../lib/gcc/x86_64-w64-mingw32/13.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: CMakeFiles\g2oCurveFitting.dir/objects.a(g2oCurveFitting.cpp.obj):g2oCurveFitting.cpp:(.text$_ZN3g2o15BlockSolverBaseC2Ev[_ZN3g2o15BlockSolverBaseC2Ev]+0x14): undefined reference to `g2o::Solver::Solver()'
C:/msys64/ucrt64/bin/../lib/gcc/x86_64-w64-mingw32/13.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: CMakeFiles\g2oCurveFitting.dir/objects.a(g2oCurveFitting.cpp.obj):g2oCurveFitting.cpp:(.text$_ZN3g2o8BaseEdgeILi1EdEC2Ev[_ZN3g2o8BaseEdgeILi1EdEC2Ev]+0x17): undefined reference to `g2o::OptimizableGraph::Edge::Edge()'
C:/msys64/ucrt64/bin/../lib/gcc/x86_64-w64-mingw32/13.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: CMakeFiles\g2oCurveFitting.dir/objects.a(g2oCurveFitting.cpp.obj):g2oCurveFitting.cpp:(.text$_ZN3g2o8BaseEdgeILi1EdEC2Ev[_ZN3g2o8BaseEdgeILi1EdEC2Ev]+0x74): undefined reference to `g2o::OptimizableGraph::Edge::~Edge()'
C:/msys64/ucrt64/bin/../lib/gcc/x86_64-w64-mingw32/13.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: CMakeFiles\g2oCurveFitting.dir/objects.a(g2oCurveFitting.cpp.obj):g2oCurveFitting.cpp:(.rdata$_ZTV16CurveFittingEdge[_ZTV16CurveFittingEdge]+0x40): undefined reference to `g2o::OptimizableGraph::Edge::setMeasurementData(double const*)'
C:/msys64/ucrt64/bin/../lib/gcc/x86_64-w64-mingw32/13.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: CMakeFiles\g2oCurveFitting.dir/objects.a(g2oCurveFitting.cpp.obj):g2oCurveFitting.cpp:(.rdata$_ZTV16CurveFittingEdge[_ZTV16CurveFittingEdge]+0x48): undefined reference to `g2o::OptimizableGraph::Edge::getMeasurementData(double*) const'
C:/msys64/ucrt64/bin/../lib/gcc/x86_64-w64-mingw32/13.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: CMakeFiles\g2oCurveFitting.dir/objects.a(g2oCurveFitting.cpp.obj):g2oCurveFitting.cpp:(.rdata$_ZTV16CurveFittingEdge[_ZTV16CurveFittingEdge]+0x50): undefined reference to `g2o::OptimizableGraph::Edge::measurementDimension() const'
C:/msys64/ucrt64/bin/../lib/gcc/x86_64-w64-mingw32/13.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: CMakeFiles\g2oCurveFitting.dir/objects.a(g2oCurveFitting.cpp.obj):g2oCurveFitting.cpp:(.rdata$_ZTV16CurveFittingEdge[_ZTV16CurveFittingEdge]+0x58): undefined reference to `g2o::OptimizableGraph::Edge::setMeasurementFromState()'
C:/msys64/ucrt64/bin/../lib/gcc/x86_64-w64-mingw32/13.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: CMakeFiles\g2oCurveFitting.dir/objects.a(g2oCurveFitting.cpp.obj):g2oCurveFitting.cpp:(.rdata$_ZTV16CurveFittingEdge[_ZTV16CurveFittingEdge]+0xd8): undefined reference to `g2o::OptimizableGraph::Edge::resolveCaches()'
C:/msys64/ucrt64/bin/../lib/gcc/x86_64-w64-mingw32/13.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: CMakeFiles\g2oCurveFitting.dir/objects.a(g2oCurveFitting.cpp.obj):g2oCurveFitting.cpp:(.rdata$_ZTVN3g2o13BaseUnaryEdgeILi1Ed18CurveFittingVertexEE[_ZTVN3g2o13BaseUnaryEdgeILi1Ed18CurveFittingVertexEE]+0x40): undefined reference to `g2o::OptimizableGraph::Edge::setMeasurementData(double const*)'
C:/msys64/ucrt64/bin/../lib/gcc/x86_64-w64-mingw32/13.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: CMakeFiles\g2oCurveFitting.dir/objects.a(g2oCurveFitting.cpp.obj):g2oCurveFitting.cpp:(.rdata$_ZTVN3g2o13BaseUnaryEdgeILi1Ed18CurveFittingVertexEE[_ZTVN3g2o13BaseUnaryEdgeILi1Ed18CurveFittingVertexEE]+0x48):  [... cut ...]
C:/msys64/ucrt64/bin/../lib/gcc/x86_64-w64-mingw32/13.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: CMakeFiles\g2oCurveFitting.dir/objects.a(g2oCurveFitting.cpp.obj):g2oCurveFitting.cpp:(.rdata$.refptr._ZN3g2o18G2OBatchStatistics12_globalStatsE[.refptr._ZN3g2o18G2OBatchStatistics12_globalStatsE]+0x0): undefined reference to `g2o::G2OBatchStatistics::_globalStats'
collect2.exe: error: ld returned 1 exit status
mingw32-make[2]: *** [CMakeFiles\g2oCurveFitting.dir\build.make:117: g2oCurveFitting.exe] Error 1
mingw32-make[1]: *** [CMakeFiles\Makefile2:86: CMakeFiles/g2oCurveFitting.dir/all] Error 2
mingw32-make: *** [Makefile:90: all] Error 2

RainerKuemmerle, 2024-12-28:
This seems more like a linking issue. Are you linking the libraries needed here?
How is your cmake setup? Can you please provide a [minimal reproducible example](https://en.wikipedia.org/wiki/Minimal_reproducible_example). Thank you.

eduardoavila1m, 2024-12-28:
Thanks so much for your response. Bellow you can find the cmake file I am using. I have successfully compiled OpenCV, Ceres and the other libraries but seem to be having this issue g2o. I've tried compiling both with c++14 and c++17. Both resulting in an error regarding the libraries.

cmake_minimum_required(VERSION 3.5)

set(CMAKE_CXX_STANDARD 14)
set(CMAKE_CXX_STANDARD_REQUIRED ON)
set(CMAKE_CXX_EXTENSIONS ON)

message("Your C++ compiler supports these C++ features:")
foreach(i ${CMAKE_CXX_COMPILE_FEATURES})
  message("${i}")
endforeach()

project(ch6)

include_directories(${Sophus_INCLUDE_DIRS})

include_directories("C:/msys64/mingw64/include/eigen3" )
include_directories("C:/msys64/mingw64/Suitesparse/eigen3" )
include_directories("C:/Program Files (x86)/Pangolin/include" )
include_directories("C:/Program Files (x86)/Pangolin/lib" )
include_directories("C:/Program Files (x86)/Pangolin/bin")
include_directories("C:/Program Files (x86)/glew/include" )
include_directories("C:/Program Files (x86)/Sophus/include" )
include_directories("C:/Program Files (x86)/Sophus/bin")
include_directories("C:/Program Files (x86)/Sophus/lib" )
include_directories("C:/opencv-4.10.0/opencv-4.10.0/Build/lib")
include_directories("C:/opencv-4.10.0/opencv-4.10.0/Build/install/include")
include_directories("C:/opencv-4.10.0/opencv-4.10.0/Build/bin")
include_directories("C:/opencv-4.10.0/opencv-4.10.0/Build")
include_directories("C:/Program Files (x86)/Ceres/include/ceres")
include_directories("C:/Program Files (x86)/Ceres/include/gmock")
include_directories("C:/Program Files (x86)/Ceres/lib/cmake/GTest")
include_directories("C:/Program Files (x86)/Ceres/lib")
include_directories("C:/Program Files (x86)/Ceres/lib/cmake/Ceres")
include_directories("C:/suitesparse-metis-for-windows-7.5.1-1/build/install/include")
include_directories("C:/suitesparse-metis-for-windows-7.5.1-1/build/install/bin")
include_directories("C:/suitesparse-metis-for-windows-7.5.1-1/build/install/lib")
include_directories("C:/suitesparse-metis-for-windows-7.5.1-1/build/install/cmake/suitesparse-7.5.1")
include_directories("C:/msys64/mingw64/include/boost")
include_directories("C:/Program Files (x86)/absl/include")
include_directories("C:/Program Files (x86)/absl/lib")
include_directories("C:/Program Files (x86)/absl/lib/cmake/absl")
include_directories("C:/Program Files (x86)/googletest-distribution/include")
include_directories("C:/Program Files (x86)/googletest-distribution/lib")
include_directories("C:/Program Files (x86)/g2o/lib/cmake/g2o")
include_directories("C:/Program Files (x86)/g2o/include")
include_directories("C:/Program Files (x86)/g2o/include/g2o")
include_directories("C:/Program Files (x86)/g2o/bin")
include_directories("C:/Program Files (x86)/g2o/lib")

# OpenCV
find_package(OpenCV REQUIRED)
include_directories(${OpenCV_INCLUDE_DIRS})

# Ceres
#find_package(blas REQUIRED)
#find_package(Suitesparse REQUIRED)
find_package(Ceres REQUIRED PATHS "C:/Program Files (x86)/Ceres/include")
include_directories(${CERES_INCLUDE_DIRS})

# g2o

SET( G2O_ROOT "C:/Program Files (x86)/g2o/include" )
find_package(G2O REQUIRED)
include_directories(${G2O_INCLUDE_DIRS})

add_executable(g2oCurveFitting_Win g2oCurveFitting_Win.cpp)
target_link_libraries(g2oCurveFitting ${OpenCV_LIBS} ${G2O_CORE_LIBRARY} ${G2O_STUFF_LIBRARY})
target_compile_features(g2oCurveFitting_Win PRIVATE cxx_std_14)
set_target_properties(g2oCurveFitting_Win PROPERTIES
    CXX_STANDARD 14
    CXX_STANDARD_REQUIRED ON
)

eduardoavila1m, 2024-12-28:
When I change to C++14 I get the following error in the running the "cmake" command. I am using MinGW. The g2o library builds correctly and installs correctly.

cmake -B build -G "MinGW Makefiles" -D "OpenCV_DIR=C:\opencv-4.10.0\opencv-4.10.0\Build" -D "Gmock_LIBRARIES=C:/Program Files (x86)/googletest-distribution/lib" -D "absl_DIR=C:\Program Files (x86)\absl\lib\cmake\absl" -D "Suitesparse_DIR=C:\suitesparse-metis-for-windows-7.5.1-1\build\install\lib\cmake\suitesparse-7.5.1" -D "Ceres_DIR=C:\Program Files (x86)\Ceres\lib\cmake\Ceres" -D "G2O_DIR=C:\Program Files (x86)\g2o\lib\cmake\g2o"

CMake Deprecation Warning at CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 3.10 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value.  Or, use the <min>...<max> syntax
  to tell CMake that the project requires at least <min> but has been updated
  to work with policies introduced by <max> or earlier.

Your C++ compiler supports these C++ features:
-- The C compiler identification is GNU 13.2.0
-- The CXX compiler identification is GNU 13.2.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/msys64/ucrt64/bin/cc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/msys64/ucrt64/bin/c++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found OpenCV: C:/opencv-4.10.0/opencv-4.10.0/Build (found version "4.10.0")
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE
-- Found required Ceres dependency: absl version 20240116 in C:\Program Files (x86)\absl\lib\cmake\absl
-- Found required Ceres dependency: Eigen version 3.4.0 in C:/msys64/mingw64/share/eigen3/cmake
-- Found Ceres version: 2.3.0 installed in: C:/Program Files (x86)/Ceres with components: [EigenSparse, SparseLinearAlgebraLibrary, SchurSpecializations]
CMake Warning (dev) at CMakeLists.txt:66 (find_package):
  Policy CMP0074 is not set: find_package uses <PackageName>_ROOT variables.
  Run "cmake --help-policy CMP0074" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.

  CMake variable G2O_ROOT is set to:

    C:/Program Files (x86)/g2o/include

  For compatibility, CMake is ignoring the variable.
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found OpenGL: opengl32
CMake Error at CMakeLists.txt:70 (target_link_libraries):
  Cannot specify link libraries for target "g2oCurveFitting" which is not
  built by this project.

-- Configuring incomplete, errors occurred!

RainerKuemmerle, 2024-12-28:
With all those `include_directories` defined I am not sure whether cmake actually found g2o. You can check if the cmake variables are populated for your linking setup. Another idea would be to rely more on the exported targets files.
I have to admit, that it is a long time ago when I used MinGW. Might be there are also issues with the exporting of symbols.

eduardoavila1m, 2024-12-28:
Thanks so much! It worked with MinGW I was missing the linking of some libraries as suggested!

closed 2024-12-28 by eduardoavila1m