RainerKuemmerle/g2o #864: g2o_viewer fails to load graphs with high-digit IDs from GTSAM LabeledSymbol

sanderboers48, 2025-01-08:
### Problem
g2o_viewer cannot load graph files containing high-digit vertex/edge IDs generated by GTSAM's LabeledSymbol. Example IDs: 27303072740933634.
When these high-digit IDs are shortened to just their last few digits (e.g., 3634), g2o_viewer loads the graphs without issues.

### Failing vs Working Format:
```
# Current (fails in g2o_viewer):
VERTEX_SE3:QUAT 27303072740933632 0 0 0 0 0 0 1
VERTEX_SE3:QUAT 27303072740933633 1 0 0 0 0 0 1
VERTEX_SE3:QUAT 27303072740933634 2 0 0 0 0 0 1
EDGE_SE3:QUAT 27303072740933632 27303072740933633 1 0 0 0 0 0 1 100 0 0 0 0 0 100 0 0 0 0 100 0 0 0 100 0 0 100 0 100
EDGE_SE3:QUAT 27303072740933633 27303072740933634 1 0 0 0 0 0 1 100 0 0 0 0 0 100 0 0 0 0 100 0 0 0 100 0 0 100 0 100

# Desired (works in g2o_viewer):
VERTEX_SE3:QUAT 3632 0 0 0 0 0 0 1
VERTEX_SE3:QUAT 3633 1 0 0 0 0 0 1
VERTEX_SE3:QUAT 3634 2 0 0 0 0 0 1
EDGE_SE3:QUAT 3632 3633 1 0 0 0 0 0 1 100 0 0 0 0 0 100 0 0 0 0 100 0 0 0 100 0 0 100 0 100
EDGE_SE3:QUAT 3633 3634 1 0 0 0 0 0 1 100 0 0 0 0 0 100 0 0 0 0 100 0 0 0 100 0 0 100 0 100
```

Does anybody know why this is an issue? And if it can be fixed?

RainerKuemmerle, 2025-01-08:
Do you have under control which keys are assigned to the `LabeledSymbol`?
Currently, g2o has `ID` modeled as `int` which typically is a 32 bit signed integer value (standard requires it to be 16-bit only).

Quickly googling, GTSAM seems to have a std::uint64_t as key. If you have under control that the key stays in the range of 32bit the files should be working.

The issue in g2o code is among others
* https://github.com/RainerKuemmerle/g2o/blob/master/g2o/core/hyper_graph.h#L163
* https://github.com/RainerKuemmerle/g2o/blob/master/g2o/core/hyper_graph.h#L144
* https://github.com/RainerKuemmerle/g2o/blob/master/g2o/core/optimizable_graph.cpp#L432

In particular the last line sets fail bits of the stream and we fail to parse also the remainder of the line.

Fixing this in g2o would require to introduce some typedef and use this consistently in the code.

Would it be an option for you to investigate controlling the range of Keys assigned to the `LabeledSymbol`?
Alternatively, if the viewer only serves as a debug tool to visualize a small Python script could re-assign numbers in the supported 32-bit range.

sanderboers48, 2025-01-09:
Thank you for your clear and helpful response! I opted to write my own function to generate a 32-bit key, which resolved the issue.  Your detailed references to the g2o code were very insightful—much appreciated!

closed 2025-01-09 by sanderboers48