RainerKuemmerle/g2o #867: What the f**k is this CMakeError?

ArdEngineer, 2025-02-05:
/usr/bin/ld: CMakeFiles/test.dir/test.cpp.o: in function `main':
test.cpp:(.text+0x278): undefined reference to `g2o::SparseOptimizer::SparseOptimizer()'
/usr/bin/ld: test.cpp:(.text+0x2b8): undefined reference to `g2o::OptimizationAlgorithmFactory::instance()'
/usr/bin/ld: test.cpp:(.text+0x2ee): undefined reference to `g2o::OptimizationAlgorithmFactory::construct(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, g2o::OptimizationAlgorithmProperty&) const'
/usr/bin/ld: test.cpp:(.text+0x30e): undefined reference to `g2o::SparseOptimizer::setAlgorithm(g2o::OptimizationAlgorithm*)'
/usr/bin/ld: test.cpp:(.text+0x510): undefined reference to `g2o::OptimizableGraph::addEdge(g2o::OptimizableGraph::Edge*)'
/usr/bin/ld: test.cpp:(.text+0x530): undefined reference to `g2o::SparseOptimizer::initializeOptimization(int)'
/usr/bin/ld: test.cpp:(.text+0x53f): undefined reference to `g2o::SparseOptimizer::computeInitialGuess()'
/usr/bin/ld: test.cpp:(.text+0x54e): undefined reference to `g2o::SparseOptimizer::computeActiveErrors()'
/usr/bin/ld: test.cpp:(.text+0x562): undefined reference to `g2o::SparseOptimizer::setVerbose(bool)'
/usr/bin/ld: test.cpp:(.text+0x57e): undefined reference to `g2o::SparseOptimizer::optimize(int, bool)'
/usr/bin/ld: test.cpp:(.text+0x67d): undefined reference to `g2o::SparseOptimizer::~SparseOptimizer()'
/usr/bin/ld: test.cpp:(.text+0x737): undefined reference to `g2o::SparseOptimizer::~SparseOptimizer()'
/usr/bin/ld: CMakeFiles/test.dir/test.cpp.o: in function `__static_initialization_and_destruction_0(int, int)':
test.cpp:(.text+0x7a6): undefined reference to `g2o_optimization_library_pcg'
/usr/bin/ld: test.cpp:(.text+0x7bc): undefined reference to `g2o_optimization_library_cholmod'
/usr/bin/ld: test.cpp:(.text+0x7d2): undefined reference to `g2o_optimization_library_csparse'
/usr/bin/ld: CMakeFiles/test.dir/test.cpp.o: in function `g2o::OptimizableGraph::addVertex(g2o::OptimizableGraph::Vertex*)':
test.cpp:(.text._ZN3g2o16OptimizableGraph9addVertexEPNS0_6VertexE[_ZN3g2o16OptimizableGraph9addVertexEPNS0_6VertexE]+0x28): undefined reference to `g2o::OptimizableGraph::addVertex(g2o::OptimizableGraph::Vertex*, g2o::HyperGraph::Data*)'
/usr/bin/ld: CMakeFiles/test.dir/test.cpp.o: in function `ResidualEdge::computeError()':
test.cpp:(.text._ZN12ResidualEdge12computeErrorEv[_ZN12ResidualEdge12computeErrorEv]+0x36): undefined reference to `typeinfo for g2o::HyperGraph::Vertex'
/usr/bin/ld: CMakeFiles/test.dir/test.cpp.o: in function `g2o::BaseVertex<3, Eigen::Matrix<double, 3, 1, 0, 3, 1> >::~BaseVertex()':
test.cpp:(.text._ZN3g2o10BaseVertexILi3EN5Eigen6MatrixIdLi3ELi1ELi0ELi3ELi1EEEED2Ev[_ZN3g2o10BaseVertexILi3EN5Eigen6MatrixIdLi3ELi1ELi0ELi3ELi1EEEED5Ev]+0x47): undefined reference to `g2o::OptimizableGraph::Vertex::~Vertex()'
/usr/bin/ld: CMakeFiles/test.dir/test.cpp.o: in function `g2o::BaseVertex<3, Eigen::Matrix<double, 3, 1, 0, 3, 1> >::BaseVertex()':
test.cpp:(.text._ZN3g2o10BaseVertexILi3EN5Eigen6MatrixIdLi3ELi1ELi0ELi3ELi1EEEEC2Ev[_ZN3g2o10BaseVertexILi3EN5Eigen6MatrixIdLi3ELi1ELi0ELi3ELi1EEEEC5Ev]+0x28): undefined reference to `g2o::OptimizableGraph::Vertex::Vertex()'
/usr/bin/ld: test.cpp:(.text._ZN3g2o10BaseVertexILi3EN5Eigen6MatrixIdLi3ELi1ELi0ELi3ELi1EEEEC2Ev[_ZN3g2o10BaseVertexILi3EN5Eigen6MatrixIdLi3ELi1ELi0ELi3ELi1EEEEC5Ev]+0xe0): undefined reference to `g2o::OptimizableGraph::Vertex::~Vertex()'
/usr/bin/ld: CMakeFiles/test.dir/test.cpp.o: in function `g2o::BaseEdge<1, Eigen::Matrix<double, 2, 1, 0, 2, 1> >::~BaseEdge()':
test.cpp:(.text._ZN3g2o8BaseEdgeILi1EN5Eigen6MatrixIdLi2ELi1ELi0ELi2ELi1EEEED2Ev[_ZN3g2o8BaseEdgeILi1EN5Eigen6MatrixIdLi2ELi1ELi0ELi2ELi1EEEED5Ev]+0x35): undefined reference to `g2o::OptimizableGraph::Edge::~Edge()'
/usr/bin/ld: CMakeFiles/test.dir/test.cpp.o: in function `g2o::BaseEdge<1, Eigen::Matrix<double, 2, 1, 0, 2, 1> >::BaseEdge()':
test.cpp:(.text._ZN3g2o8BaseEdgeILi1EN5Eigen6MatrixIdLi2ELi1ELi0ELi2ELi1EEEEC2Ev[_ZN3g2o8BaseEdgeILi1EN5Eigen6MatrixIdLi2ELi1ELi0ELi2ELi1EEEEC5Ev]+0x19): undefined reference to `g2o::OptimizableGraph::Edge::Edge()'
/usr/bin/ld: test.cpp:(.text._ZN3g2o8BaseEdgeILi1EN5Eigen6MatrixIdLi2ELi1ELi0ELi2ELi1EEEEC2Ev[_ZN3g2o8BaseEdgeILi1EN5Eigen6MatrixIdLi2ELi1ELi0ELi2ELi1EEEEC5Ev]+0x8c): undefined reference to `g2o::OptimizableGraph::Edge::~Edge()'
/usr/bin/ld: CMakeFiles/test.dir/test.cpp.o:(.data.rel.ro._ZTV12ResidualEdge[_ZTV12ResidualEdge]+0x40): undefined reference to `g2o::OptimizableGraph::Edge::setMeasurementData(double const*)'
/usr/bin/ld: CMakeFiles/test.dir/test.cpp.o:(.data.rel.ro._ZTV12ResidualEdge[_ZTV12ResidualEdge]+0x48): undefined reference to `g2o::OptimizableGraph::Edge::getMeasurementData(double*) const'
/usr/bin/ld: CMakeFiles/test.dir/test.cpp.o:(.data.rel.ro._ZTV12ResidualEdge[_ZTV12ResidualEdge]+0x50): undefined reference to `g2o::OptimizableGraph::Edge::measurementDimension() const'
/usr/bin/ld: CMakeFiles/test.dir/test.cpp.o:(.data.rel.ro._ZTV12ResidualEdge[_ZTV12ResidualEdge]+0x58): undefined reference to `g2o::OptimizableGraph::Edge::setMeasurementFromState()'
/usr/bin/ld: CMakeFiles/test.dir/test.cpp.o:(.data.rel.ro._ZTV12ResidualEdge[_ZTV12ResidualEdge]+0xd8): undefined reference to `g2o::OptimizableGraph::Edge::resolveCaches()'
/usr/bin/ld: CMakeFiles/test.dir/test.cpp.o:(.data.rel.ro._ZTVN3g2o13BaseUnaryEdgeILi1EN5Eigen6MatrixIdLi2ELi1ELi0ELi2ELi1EEE12VertexParamsEE[_ZTVN3g2o13BaseUnaryEdgeILi1EN5Eigen6MatrixIdLi2ELi1ELi0ELi2ELi1EEE12VertexParamsEE]+0x40): undefined reference to `g2o::OptimizableGraph::Edge::setMeasurementData(double const*)'
/usr/bin/ld: CMakeFiles/test.dir/test.cpp.o:(.data.rel.ro._ZTVN3g2o13BaseUnaryEdgeILi1EN5Eigen6MatrixIdLi2ELi1ELi0ELi2ELi1EEE12VertexParamsEE[_ZTVN3g2o13BaseUnaryEdgeILi1EN5Eigen6MatrixIdLi2ELi1ELi0ELi2ELi1EEE12VertexParamsEE]+0x48): undefined reference to `g2o::OptimizableGraph::Edge::getMeasurementData(double*) const'
/usr/bin/ld: CMakeFiles/test.dir/test.cpp.o:(.data.rel.ro._ZTVN3g2o13BaseUnaryEdgeILi1EN5Eigen6MatrixIdLi2ELi1ELi0ELi2ELi1EEE12VertexParamsEE[_ZTVN3g2o13BaseUnaryEdgeILi1EN5Eigen6MatrixIdLi2ELi1ELi0ELi2ELi1EEE12VertexParamsEE]+0x50): undefined reference to `g2o::OptimizableGraph::Edge::measurementDimension() const'
/usr/bin/ld: CMakeFiles/test.dir/test.cpp.o:(.data.rel.ro._ZTVN3g2o13BaseUnaryEdgeILi1EN5Eigen6MatrixIdLi2ELi1ELi0ELi2ELi1EEE12VertexParamsEE[_ZTVN3g2o13BaseUnaryEdgeILi1EN5Eigen6MatrixIdLi2ELi1ELi0ELi2ELi1EEE12VertexParamsEE]+0x58): undefined reference to `g2o::OptimizableGraph::Edge::setMeasurementFromState()'
/usr/bin/ld: CMakeFiles/test.dir/test.cpp.o:(.data.rel.ro._ZTVN3g2o13BaseUnaryEdgeILi1EN5Eigen6MatrixIdLi2ELi1ELi0ELi2ELi1EEE12VertexParamsEE[_ZTVN3g2o13BaseUnaryEdgeILi1EN5Eigen6MatrixIdLi2ELi1ELi0ELi2ELi1EEE12VertexParamsEE]+0xd8): undefined reference to `g2o::OptimizableGraph::Edge::resolveCaches()'
/usr/bin/ld: CMakeFiles/test.dir/test.cpp.o:(.data.rel.ro._ZTVN3g2o18BaseFixedSizedEdgeILi1EN5Eigen6MatrixIdLi2ELi1ELi0ELi2ELi1EEEJ12VertexParamsEEE[_ZTVN3g2o18BaseFixedSizedEdgeILi1EN5Eigen6MatrixIdLi2ELi1ELi0ELi2ELi1EEEJ12VertexParamsEEE]+0x40): undefined reference to `g2o::OptimizableGraph::Edge::setMeasurementData(double const*)'
/usr/bin/ld: CMakeFiles/test.dir/test.cpp.o:(.data.rel.ro._ZTVN3g2o18BaseFixedSizedEdgeILi1EN5Eigen6MatrixIdLi2ELi1ELi0ELi2ELi1EEEJ12VertexParamsEEE[_ZTVN3g2o18BaseFixedSizedEdgeILi1EN5Eigen6MatrixIdLi2ELi1ELi0ELi2ELi1EEEJ12VertexParamsEEE]+0x48): undefined reference to `g2o::OptimizableGraph::Edge::getMeasurementData(double*) const'
/usr/bin/ld: CMakeFiles/test.dir/test.cpp.o:(.data.rel.ro._ZTVN3g2o18BaseFixedSizedEdgeILi1EN5Eigen6MatrixIdLi2ELi1ELi0ELi2ELi1EEEJ12VertexParamsEEE[_ZTVN3g2o18BaseFixedSizedEdgeILi1EN5Eigen6MatrixIdLi2ELi1ELi0ELi2ELi1EEEJ12VertexParamsEEE]+0x50): undefined reference to `g2o::OptimizableGraph::Edge::measurementDimension() const'
/usr/bin/ld: CMakeFiles/test.dir/test.cpp.o:(.data.rel.ro._ZTVN3g2o18BaseFixedSizedEdgeILi1EN5Eigen6MatrixIdLi2ELi1ELi0ELi2ELi1EEEJ12VertexParamsEEE[_ZTVN3g2o18BaseFixedSizedEdgeILi1EN5Eigen6MatrixIdLi2ELi1ELi0ELi2ELi1EEEJ12VertexParamsEEE]+0x58): undefined reference to `g2o::OptimizableGraph::Edge::setMeasurementFromState()'
/usr/bin/ld: CMakeFiles/test.dir/test.cpp.o:(.data.rel.ro._ZTVN3g2o18BaseFixedSizedEdgeILi1EN5Eigen6MatrixIdLi2ELi1ELi0ELi2ELi1EEEJ12VertexParamsEEE[_ZTVN3g2o18BaseFixedSizedEdgeILi1EN5Eigen6MatrixIdLi2ELi1ELi0ELi2ELi1EEEJ12VertexParamsEEE]+0xd8): undefined reference to `g2o::OptimizableGraph::Edge::resolveCaches()'
[... cut ...]
/usr/bin/ld: CMakeFiles/test.dir/test.cpp.o: in function `g2o::BaseFixedSizedEdge<1, Eigen::Matrix<double, 2, 1, 0, 2, 1>, VertexParams>::resize(unsigned long)':
test.cpp:(.text._ZN3g2o18BaseFixedSizedEdgeILi1EN5Eigen6MatrixIdLi2ELi1ELi0ELi2ELi1EEEJ12VertexParamsEE6resizeEm[_ZN3g2o18BaseFixedSizedEdgeILi1EN5Eigen6MatrixIdLi2ELi1ELi0ELi2ELi1EEEJ12VertexParamsEE6resizeEm]+0x49): undefined reference to `g2o::HyperGraph::Edge::resize(unsigned long)'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/test.dir/build.make:100: test] Error 1
make[1]: *** [CMakeFiles/Makefile2:87: CMakeFiles/test.dir/all] Error 2

ArdEngineer, 2025-02-05:
Here is my CMake file, what's going on? It seems that it's not the problem of cxx standard

[image]

RainerKuemmerle, 2025-02-05:
Your linker config seems not correct.
Try debug if the variables like `${G2O_CORE_LIBRARY}` are actually set.
Instead of using cmake's Module way of finding g2o, try the Config mode to use the exported targets.
https://cmake.org/cmake/help/book/mastering-cmake/chapter/Finding%20Packages.html

ArdEngineer, 2025-02-09:
> Your linker config seems not correct. Try debug if the variables like `${G2O_CORE_LIBRARY}` are actually set. Instead of using cmake's Module way of finding g2o, try the Config mode to use the exported targets. https://cmake.org/cmake/help/book/mastering-cmake/chapter/Finding%20Packages.html

Thanks for yor help! I have solve this problem.

ArdEngineer, 2025-02-09:
> Your linker config seems not correct. Try debug if the variables like `${G2O_CORE_LIBRARY}` are actually set. Instead of using cmake's Module way of finding g2o, try the Config mode to use the exported targets. https://cmake.org/cmake/help/book/mastering-cmake/chapter/Finding%20Packages.html

Thanks for your help! I have solve this problem.

closed 2025-03-22 by RainerKuemmerle