RainerKuemmerle/g2o #868: Information matrix and weighing behaviour

SteveDoingStuff, 2025-02-07:
Hi all!
First of all thanks Rainer for the amazing work by providing your framework for pose graph optimization!

I'm using your library in my master thesis for optimizing my graph consisting of scan poses as nodes and GICP derived relative poses as edges. As information matrices I use the hessians of the GICP result (using small_gicp). Edges linking two poses that are further apart naturally have lower hessian values (or higher covariances) than 'shorter' edges due to less overlap and lower accuracy for points farther away from the scanner. The object scanned is the exterior and interior of a sizable mansion. When applying the corresponding hessians (GICP) in edge->setInformation(), I get high residuals on certain edges with low hessian values, which is to be expected but:

When I check the point clouds using their optimized poses, I see significant misalignments (in a magnitude of around 30cm) especially on edges with high covariance. Even though the GICP result was visually very accurate, the optimization distorts these edges because of their high covariance.

When use equal weight (Identity matrix) as information on all edges I get nice alignments after the optimization but the scale of the covariances of the optimized poses seems to be only relative. 

Is it justified to use for example some sort of median hessian as information matrix on all edges (to get realistic scaling)? I tried this and also get the desired alignments on all edges.

When I use the Identity matrix for all edges, how can I scale the resulting covariances of the poses? How can I 'tell' g2o that im working with meters as units, while applying equal weight to all edges? Most importantly if I'm not overlooking something here, is there some literature adressing this 'problem'?

Thank you,
Stefan

RainerKuemmerle, 2025-02-09:
Did you have a look whether the error function of the g2o edges you use aligns with the hessian obtained by the ICP algorithm you are using?
The default 3D pose constraint in g2o uses a Quaternion subspace error space. Probably something the ICP algorithm does not use.

Other than that, estimating the covariance from scan matching can range from heuristics, e.g., number of correspondences (which might be a good approximation to start) or more sophisticated approaches which more closely try to model the characteristics of the range sensor - those probably have then to be modeled as fitting to your specific sensor.
When I was last working with 3D maps, I noticed that orientation is typically estimated well on good matches and translation is more important to be adjusted on smoothing.

SteveDoingStuff, 2025-02-09:
I now followed up on your suggestion to use heuristic data, and used Identity matrices, scaled by the cumulative error of the GICP result divided by the number of correspondences. I get very good alignments, the residuals of the edges are of the same population the scanner accuracy data would suggest. Seems like I over complicated things with the hessian (though the hessian of the GICP result should align with the error function). I found some other resources to estimate ICP uncertainty (Brossard, M., Bonnabel, S., & Barrau, A. (2020). A new approach to 3D ICP covariance estimation) but we deemed it out of scope for my thesis for now. 

Thank you for the advice!

closed 2025-04-25 by github-actions[bot]