RainerKuemmerle/g2o #871: Considerations for creating g2o files

TakShimoda, 2025-02-11:
Hello, thank you for this library.

I'm currently trying to create a simple pose graph file in g2o format where a ground robot accumulates drift by moving in a 0.5mx0.5m square pattern about 15 times. It gets odometry from its 20Hz 6DoF IMU and wheel encoder, and measurements are taken by observing a stationary AprilTag by a realsense D435i as measurement constraints for non-consecutive poses. Below is an example of the odometry accumulating drift after many turns, compared to ground-truth (gt) captured by a motion capture system:

[image]

Right now, I'm trying to create a simple g2o file for quick optimization with the default GN and LM optimizers from g2o-viewer to see if the file is correctly made for optimization problems. I have used references from sources such as [here](https://github.com/UditSinghParihar/g2o_tutorial) and the 2D datasets from  [here](https://lucacarlone.mit.edu/datasets/). I have attached an example file I created below (with .txt extension as github isnt' allowing me to attach a .g2o file) which is in the [2D-SLAM format with vertices along with odom and loop closure edges](https://github.com/RainerKuemmerle/g2o/wiki/File-Format-SLAM-2D.)

[input_2D.txt](https://github.com/user-attachments/files/18758001/input_2D.txt)

What I did to create the file was the following:
1) Downsample the odometry measurements as it's originally 20Hz with a robot that moves relatively slow (only up to 0.18m/s linearly, and angular velocity of 0.18rad/s). I only took every 4th odom measurement as if it was at 5Hz. I noticed the graph looked very dense with nearby poses which might have been poor for optimization.
2) Relate an observation of an AprilTag with an odometry pose with the nearest timestamp, given they're within 0.01 second of eachother. This approximately relates a vertex with a measurement to create loop closure edges. I have already taken care of processing steps such as converting measurements to the same coordinate system (ROS in my case) and checking the measurements were correct and accurate.
3) Create odometry and loop closure edges. For odometry between poses i and j, for pose j expressed in pose i, I did
$T^i_{j} = (T^W_{i})^{-1}  T^W_{j} = T^i_{W}  T^W_{j} \in \mathrm{SE}(2)$, where W is the world-frame and we're expressing the pose of the subscript frame in the frame of the superscript frame. For measurements, given two poses i and j observing a landmark L, which is expressed in their own frames, I calculate the measurement $z_{ij}$  between them with $z_{ij} = T^i_{L} (T^j_{L})^{-1} = T^i_{L}  T^L_{j} \in \mathrm{SE}(2)$
4) Limit the number of measurement edges. This is because I want the measurements to only correct loop closures after drift, but as seen by the odometry poses, it's fairly accurate until it loops around at least 10 times. So for poses i and j, I make sure to only make measurement edges if they're at least a certain index range apart (e.g. between 30%-70% of n apart from each other if there are n poses) for step 3 mentioned above.
5) Tune the information matrix. This is where I feel it's affecting the solution. The odometry and AprilTag measurements come from ROS libraries and they give 0 covariances. From the two g2o sources I was referencing above, I started with a heuristic to keep the values of $I_{xx}$, $I_{yy}$, and $I_{\theta\theta}$ constant and keeping the off-diagonals zero. I initially set it to around (5, 5, 10) for odom and (7, 7, 10) for measurements, considering the magnitude of my trajectory is fairly small. I also tried scaling the diagonal entries to be inversely proportional to the scale of the relative measurement between two poses, and made the range large for theta (from around 100-5000) as I noticed this was the case for the Intel 2D dataset. 

The best result I've gotten so far visually looks like:

[image]

Without going into too much details, especially the front-end, do you have suggestions on modifying this file to improve the results with basic optimizers? I've been trying to adjust each of the factors I mentioned above, but the results don't seem to improve much. I'm using the default gn_var_cholmod optimizer but all the other ones essentially have the same behavior. Are there more details in the literature pertaining to properly modelling this setup, such as how to properly model uncertainty given the specific measurement models, and how dense these factor graphs should be? I've looked at the g2o papers/documents and the other pose graph datasets available but they don't divulge too much information on these specifics.

Thanks!

RainerKuemmerle, 2025-02-22:
I am not exactly sure I can follow how you determine the loop closure constraints.
Maybe it makes sense to verify measurements individually while still knowing the measurements for L in both frames.
I assume you take into account static transformations between odom frame and the camera observing the landmark.

For example, inspect if between poses far in time the transformation is as you calculate it. Visualize where the poses i and j see the landmark. Also to understand how much noise is in your system.

Alternatively, you can think of directly adding the relative observation of the AprilTag as measurement.
For example as done in this publicly available data: https://www.informatik.uni-bremen.de/agebv/en/DlrSpatialCognitionDataSet

Data Set without images (12MB)

The folder `/dlr-spatial_cognition_data/dlr-spatial_cognition-e/` contains ground truth data associations of the landmarks  (in your case the April Tag) and a description of the format.

Below this data converted into g2o's format. You can see how the edges at the end of the file correspond to the content of map1_0.circles of the data set.
[dlr.zip](https://github.com/user-attachments/files/18923157/dlr.zip)

As you have a 2D pose of the tag, another alternative is to add the tag also as a "pose" and just add the observations directly.

TakShimoda, 2025-02-26:
Hi Rainer, 

For my loop closure constraints, I just have the camera observe the same stationary AprilTag(s) with known ID (so I already know the data association) at the same place at different time steps (e.g. after the robot returns to the same spot to see the same tag after finishing a lap). 
I did double check my measurements and they were fairly accurate as the AprilTag3 library I used is robust enough for the distances and orientations I observe. I even used vicon markers on top of the AprilTags as ground-truth markers so I can check the accuracy of the measurements and they were reasonable. 

I'll try to make my file like the Spatial Cognition DataSet and directly add the AprilTag measurements as landmarks. I just had questions about the format of the g2o file you provided:
1) Do VERTEX_XY specify landmark positions in a fixed global frame (e.g. the fixed node) or the most recent odometry frame? For the EDGE_SE2_XY entries, I see those edges correspond to LANDMARK_C entries in the map1_0.circles which are the positions of the circular landmarks in the robot frame, so I'm assuming those vertices are obtained by converting the reference frame of the landmarks from the robot odom frame at time step i to a global frame.
2) I counted 584 unique circle marker IDs. Do each of these only appear once in VERTEX_XY, and is that entry based on the first frame that observed them (since one landmark will be observed in multiple frames)? So in my case, if my data had 4 AprilTags, will there only be 4 VERTEX_XY entries? 
3) What is the format of the EDGE_SE2_XY entries? I don't see it on the wiki but I'm assuming it's `i j x y Ixx Ixy Iyy`
4) I'm assuming the numbering of the ID doesn't matter. Odom doesn't seem to be consecutive in the data, and it seems certain indices are skipped. The vertex index goes up to 17596 even though there are only 3875 vertices.

Thanks

closed 2025-05-12 by github-actions[bot]