RainerKuemmerle/g2o #873: Python Embeddings: cmake error

aadityapattabhiraman, 2025-02-24:
I am trying to install python embedding for g2o

---

What i ran: 
```bash
git clone -b pymem --single-branch https://github.com/RainerKuemmerle/g2o
cd g2o
mkdir build
cd build
cmake -DG2O_BUILD_PYTHON=ON ..
``` 

I expected it to finish build without any errors as i have done it just a few days back.  

---

What i get:
```bash
-- The C compiler identification is GNU 14.2.1
-- The CXX compiler identification is GNU 14.2.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Compiling on Unix
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE
-- Failed to find SuiteSparse - Did not find BLAS library (required for SuiteSparse).
-- Failed to find SuiteSparse - Did not find LAPACK library (required for SuiteSparse).
-- Failed to find SuiteSparse - Did not find AMD header (required SuiteSparse component).
-- Failed to find SuiteSparse - Did not find AMD library (required SuiteSparse component).
-- Failed to find SuiteSparse - Did not find CAMD header (required SuiteSparse component).
-- Failed to find SuiteSparse - Did not find CAMD library (required SuiteSparse component).
-- Failed to find SuiteSparse - Did not find CCOLAMD header (required SuiteSparse component).
-- Failed to find SuiteSparse - Did not find CCOLAMD library (required SuiteSparse component).
-- Failed to find SuiteSparse - Did not find CHOLMOD header (required SuiteSparse component).
-- Failed to find SuiteSparse - Did not find CHOLMOD library (required SuiteSparse component).
-- Failed to find SuiteSparse - Did not find COLAMD header (required SuiteSparse component).
-- Failed to find SuiteSparse - Did not find COLAMD library (required SuiteSparse component).
-- Failed to find SuiteSparse - Did not find SPQR header (required SuiteSparse component).
-- Failed to find SuiteSparse - Did not find SPQR library (required SuiteSparse component).
-- Failed to find SuiteSparse - Did not find Config header (required SuiteSparse component).
-- Failed to find SuiteSparse - Did not find Config library (required SuiteSparse component).
-- Looking for shm_open in rt
-- Looking for shm_open in rt - found
-- Failed to find some/all required components of SuiteSparse. (missing: SuiteSparse_AMD_INCLUDE_DIR SuiteSparse_AMD_LIBRARY SuiteSparse_CAMD_INCLUDE_DIR SuiteSparse_CAMD_LIBRARY SuiteSparse_CCOLAMD_INCLUDE_DIR SuiteSparse_CCOLAMD_LIBRARY SuiteSparse_CHOLMOD_INCLUDE_DIR SuiteSparse_CHOLMOD_LIBRARY SuiteSparse_COLAMD_INCLUDE_DIR SuiteSparse_COLAMD_LIBRARY SuiteSparse_SPQR_INCLUDE_DIR SuiteSparse_SPQR_LIBRARY SuiteSparse_Config_INCLUDE_DIR SuiteSparse_Config_LIBRARY AMD CAMD CCOLAMD CHOLMOD COLAMD SPQR Config) 
-- Disable support for Cholmod
-- Building LGPL code as a shared library
-- Could NOT find CSparse (missing: CSPARSE_INCLUDE_DIR CSPARSE_LIBRARY) 
-- Disable support for CSparse
-- Found OpenGL: /usr/lib/libOpenGL.so
-- Compiling with OpenGL support
-- Could NOT find QGLViewer (missing: QGLVIEWER_INCLUDE_DIR QGLVIEWER_LIBRARY) 
-- Compiling SLAM 3D types
-- Compiling built in types slam2d slam2d (addons) data sclam2d slam3d slam3d (addons) sba icp sim3
-- Compiling g2o apps
-- Compiling g2o examples
-- Performing Test SSE2_AVAILABLE
-- Performing Test SSE2_AVAILABLE - Success
-- Performing Test SSE3_AVAILABLE
-- Performing Test SSE3_AVAILABLE - Success
-- Performing Test SSSE3_AVAILABLE
-- Performing Test SSSE3_AVAILABLE - Success
-- Performing Test SSE41_AVAILABLE
-- Performing Test SSE41_AVAILABLE - Success
-- Performing Test SSE42_AVAILABLE
-- Performing Test SSE42_AVAILABLE - Success
-- Compiling with GCC
-- Found Eigen3: /usr/include/eigen3 (Required is at least version "3.3")
CMake Warning at CMakeLists.txt:424 (find_package):
  By not providing "Findnlohmann_json.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "nlohmann_json", but CMake did not find one.

  Could not find a package configuration file provided by "nlohmann_json"
  (requested version 3.3.0) with any of the following names:

    nlohmann_jsonConfig.cmake
    nlohmann_json-config.cmake

  Add the installation prefix of "nlohmann_json" to CMAKE_PREFIX_PATH or set
  "nlohmann_json_DIR" to a directory containing one of the above files.  If
  "nlohmann_json" provides a separate development package or SDK, be sure it
  has been installed.

-- Compiling g2o's python wrapper
CMake Deprecation Warning at build/_deps/pybind11-src/CMakeLists.txt:13 (cmake_minimum_required):
  Compatibility with CMake < 3.10 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value.  Or, use the <min>...<max> syntax
  to tell CMake that the project requires at least <min> but has been updated
  to work with policies introduced by <max> or earlier.

-- pybind11 v2.13.6 
-- Found Python: /usr/bin/python3.13 (found suitable version "3.13.2", minimum required is "3.7") found components: Interpreter Development.Module Development.Embed
-- Performing Test HAS_FLTO
-- Performing Test HAS_FLTO - Success
-- Configuring done (6.3s)
CMake Error at g2o/EXTERNAL/freeglut/CMakeLists.txt:7 (target_link_libraries):
  Target "freeglut_minimal" links to:

    OpenGL::GLU

  but the target was not found.  Possible reasons include:

    * There is a typo in the target name.
    * A find_package call is missing for an IMPORTED target.
    * An ALIAS target is missing.

CMake Error at g2o/stuff/CMakeLists.txt:67 (target_link_libraries):
  Target "opengl_helper" links to:

    OpenGL::GLU

  but the target was not found.  Possible reasons include:

    * There is a typo in the target name.
    * A find_package call is missing for an IMPORTED target.
    * An ALIAS target is missing.

-- Generating done (0.1s)
CMake Generate step failed.  Build files cannot be regenerated correctly.
```

Thank you in advance

aadityapattabhiraman, 2025-02-25:
I have solved this error. I had to install glu and suitesparse along with eigen. Why not add those 2 packages as dependencies

closed 2025-02-25 by aadityapattabhiraman

RainerKuemmerle, 2025-02-25:
Suitesparse and GLU should be optional.Suitesparse I am pretty sure it works to build without being installed.GL might be broken.