RainerKuemmerle/g2o #875: Link error, multiple definitions for function

TLescoatTFX, 2025-03-18:
When compiling g2o for stella_vslam on Visual Studio 2022 on Windows 11, I get a ton of link errors:

    3>g2o_types_sba.lib(g2o_types_sba.dll) : error LNK2005: "public: __cdecl g2o::BaseVertex<6,class g2o::SE3Quat>::BaseVertex<6,class g2o::SE3Quat>(void)" (??0?$BaseVertex@$05VSE3Quat@g2o@@@g2o@@QEAA@XZ) déjà défini(e) dans stella_vslam.lib(pose_optimizer_g2o.obj)
    3>g2o_types_sba.lib(g2o_types_sba.dll) : error LNK2005: "public: virtual void __cdecl g2o::BaseVertex<6,class g2o::SE3Quat>::push(void)" (?push@?$BaseVertex@$05VSE3Quat@g2o@@@g2o@@UEAAXXZ) déjà défini(e) dans stella_vslam.lib(pose_optimizer_g2o.obj)
    3>g2o_types_sba.lib(g2o_types_sba.dll) : error LNK2005: "public: virtual void __cdecl g2o::BaseVertex<6,class g2o::SE3Quat>::pop(void)" (?pop@?$BaseVertex@$05VSE3Quat@g2o@@@g2o@@UEAAXXZ) déjà défini(e) dans stella_vslam.lib(pose_optimizer_g2o.obj)
    3>g2o_types_sba.lib(g2o_types_sba.dll) : error LNK2005: "public: virtual __cdecl g2o::BaseVertex<6,class g2o::SE3Quat>::~BaseVertex<6,class g2o::SE3Quat>(void)" (??1?$BaseVertex@$05VSE3Quat@g2o@@@g2o@@UEAA@XZ) déjà défini(e) dans stella_vslam.lib(pose_optimizer_g2o.obj)
    3>g2o_types_sba.lib(g2o_types_sba.dll) : error LNK2005: "public: virtual __cdecl g2o::BaseEdge<2,class Eigen::Matrix<double,2,1,0,2,1> >::~BaseEdge<2,class Eigen::Matrix<double,2,1,0,2,1> >(void)" (??1?$BaseEdge@$01V?$Matrix@N$01$00$0A@$01$00@Eigen@@@g2o@@UEAA@XZ) déjà défini(e) dans stella_vslam.lib(pose_optimizer_g2o.obj)
    3>g2o_types_sba.lib(g2o_types_sba.dll) : error LNK2005: "public: virtual void __cdecl g2o::BaseEdge<2,class Eigen::Matrix<double,2,1,0,2,1> >::setMeasurement(class Eigen::Matrix<double,2,1,0,2,1> const &)" (?setMeasurement@?$BaseEdge@$01V?$Matrix@N$01$00$0A@$01$00@Eigen@@@g2o@@UEAAXAEBV?$Matrix@N$01$00$0A@$01$00@Eigen@@@Z) déjà défini(e) dans stella_vslam.lib(pose_optimizer_g2o.obj)
    3>g2o_types_sba.lib(g2o_types_sba.dll) : error LNK2005: "public: virtual __cdecl g2o::BaseEdge<3,class Eigen::Matrix<double,3,1,0,3,1> >::~BaseEdge<3,class Eigen::Matrix<double,3,1,0,3,1> >(void)" (??1?$BaseEdge@$02V?$Matrix@N$02$00$0A@$02$00@Eigen@@@g2o@@UEAA@XZ) déjà défini(e) dans stella_vslam.lib(pose_optimizer_g2o.obj)
    3>g2o_types_sba.lib(g2o_types_sba.dll) : error LNK2005: "public: void __cdecl g2o::BaseEdge<3,class Eigen::Matrix<double,3,1,0,3,1> >::setInformation(class Eigen::Matrix<double,3,3,0,3,3> const &)" (?setInformation@?$BaseEdge@$02V?$Matrix@N$02$00$0A@$02$00@Eigen@@@g2o@@QEAAXAEBV?$Matrix@N$02$02$0A@$02$02@Eigen@@@Z) déjà défini(e) dans stella_vslam.lib(pose_optimizer_g2o.obj)
    3>g2o_types_sba.lib(g2o_types_sba.dll) : error LNK2005: "public: virtual void __cdecl g2o::BaseEdge<3,class Eigen::Matrix<double,3,1,0,3,1> >::setMeasurement(class Eigen::Matrix<double,3,1,0,3,1> const &)" (?setMeasurement@?$BaseEdge@$02V?$Matrix@N$02$00$0A@$02$00@Eigen@@@g2o@@UEAAXAEBV?$Matrix@N$02$00$0A@$02$00@Eigen@@@Z) déjà défini(e) dans stella_vslam.lib(pose_optimizer_g2o.obj)
    3>stella_vslam.lib(graph_optimizer.obj) : error LNK2019: symbole externe non résolu "public: __cdecl g2o::csparse::CSparse::CSparse(void)" (??0CSparse@csparse@g2o@@QEAA@XZ) référencé dans la fonction "class std::unique_ptr<class g2o::LinearSolverCSparse<class Eigen::Matrix<double,7,7,0,7,7> >,struct std::default_delete<class g2o::LinearSolverCSparse<class Eigen::Matrix<double,7,7,0,7,7> > > > __cdecl stella_vslam::make_unique<class g2o::LinearSolverCSparse<class Eigen::Matrix<double,7,7,0,7,7> > >(void)" (??$make_unique@V?$LinearSolverCSparse@V?$Matrix@N$06$06$0A@$06$06@Eigen@@@g2o@@$$V@stella_vslam@@YA?AV?$unique_ptr@V?$LinearSolverCSparse@V?$Matrix@N$06$06$0A@$06$06@Eigen@@@g2o@@U?$default_delete@V?$LinearSolverCSparse@V?$Matrix@N$06$06$0A@$06$06@Eigen@@@g2o@@@std@@@std@@XZ)
    3>stella_vslam.lib(global_bundle_adjuster.obj) : error LNK2001: symbole externe non résolu "public: __cdecl g2o::csparse::CSparse::CSparse(void)" (??0CSparse@csparse@g2o@@QEAA@XZ)
    3>stella_vslam.lib(graph_optimizer.obj) : error LNK2019: symbole externe non résolu "public: __cdecl g2o::csparse::CSparse::~CSparse(void)" (??1CSparse@csparse@g2o@@QEAA@XZ) référencé dans la fonction "int `class std::unique_ptr<class g2o::LinearSolverCSparse<class Eigen::Matrix<double,7,7,0,7,7> >,struct std::default_delete<class g2o::LinearSolverCSparse<class Eigen::Matrix<double,7,7,0,7,7> > > > __cdecl stella_vslam::make_unique<class g2o::LinearSolverCSparse<class Eigen::Matrix<double,7,7,0,7,7> > >(void)'::`1'::dtor$3" (?dtor$3@?0???$make_unique@V?$LinearSolverCSparse@V?$Matrix@N$06$06$0A@$06$06@Eigen@@@g2o@@$$V@stella_vslam@@YA?AV?$unique_ptr@V?$LinearSolverCSparse@V?$Matrix@N$06$06$0A@$06$06@Eigen@@@g2o@@U?$default_delete@V?$LinearSolverCSparse@V?$Matrix@N$06$06$0A@$06$06@Eigen@@@g2o@@@std@@@std@@XZ@4HA)
    3>stella_vslam.lib(global_bundle_adjuster.obj) : error LNK2001: symbole externe non résolu "public: __cdecl g2o::csparse::CSparse::~CSparse(void)" (??1CSparse@csparse@g2o@@QEAA@XZ)
    3>stella_vslam.lib(graph_optimizer.obj) : error LNK2019: symbole externe non résolu "public: bool __cdecl g2o::csparse::CSparse::factorize(void)" (?factorize@CSparse@csparse@g2o@@QEAA_NXZ) référencé dans la fonction "protected: virtual bool __cdecl g2o::LinearSolverCSparse<class Eigen::Matrix<double,7,7,0,7,7> >::solveBlocks_impl(class g2o::SparseBlockMatrix<class Eigen::Matrix<double,7,7,0,7,7> > const &,class std::function<void __cdecl(class g2o::MarginalCovarianceCholesky &)>)" (?solveBlocks_impl@?$LinearSolverCSparse@V?$Matrix@N$06$06$0A@$06$06@Eigen@@@g2o@@MEAA_NAEBV?$SparseBlockMatrix@V?$Matrix@N$06$06$0A@$06$06@Eigen@@@2@V?$function@$$A6AXAEAVMarginalCovarianceCholesky@g2o@@@Z@std@@@Z)
    3>stella_vslam.lib(global_bundle_adjuster.obj) : error LNK2001: symbole externe non résolu "public: bool __cdecl g2o::csparse::CSparse::factorize(void)" (?factorize@CSparse@csparse@g2o@@QEAA_NXZ)
    3>stella_vslam.lib(graph_optimizer.obj) : error LNK2019: symbole externe non résolu "public: void __cdecl g2o::csparse::CSparse::freeFactor(void)" (?freeFactor@CSparse@csparse@g2o@@QEAAXXZ) référencé dans la fonction "protected: virtual bool __cdecl g2o::LinearSolverCSparse<class Eigen::Matrix<double,7,7,0,7,7> >::solveBlocks_impl(class g2o::SparseBlockMatrix<class Eigen::Matrix<double,7,7,0,7,7> > const &,class std::function<void __cdecl(class g2o::MarginalCovarianceCholesky &)>)" (?solveBlocks_impl@?$LinearSolverCSparse@V?$Matrix@N$06$06$0A@$06$06@Eigen@@@g2o@@MEAA_NAEBV?$SparseBlockMatrix@V?$Matrix@N$06$06$0A@$06$06@Eigen@@@2@V?$function@$$A6AXAEAVMarginalCovarianceCholesky@g2o@@@Z@std@@@Z)
    3>stella_vslam.lib(global_bundle_adjuster.obj) : error LNK2001: symbole externe non résolu "public: void __cdecl g2o::csparse::CSparse::freeFactor(void)" (?freeFactor@CSparse@csparse@g2o@@QEAAXXZ)
    3>stella_vslam.lib(graph_optimizer.obj) : error LNK2019: symbole externe non résolu "public: struct g2o::csparse::CSparse::FactorView __cdecl g2o::csparse::CSparse::factor(void)" (?factor@CSparse@csparse@g2o@@QEAA?AUFactorView@123@XZ) référencé dans la fonction "protected: virtual bool __cdecl g2o::LinearSolverCSparse<class Eigen::Matrix<double,7,7,0,7,7> >::solveBlocks_impl(class g2o::SparseBlockMatrix<class Eigen::Matrix<double,7,7,0,7,7> > const &,class std::function<void __cdecl(class g2o::MarginalCovarianceCholesky &)>)" (?solveBlocks_impl@?$LinearSolverCSparse@V?$Matrix@N$06$06$0A@$06$06@Eigen@@@g2o@@MEAA_NAEBV?$SparseBlockMatrix@V?$Matrix@N$06$06$0A@$06$06@Eigen@@@2@V?$function@$$A6AXAEAVMarginalCovarianceCholesky@g2o@@@Z@std@@@Z)
    3>stella_vslam.lib(global_bundle_adjuster.obj) : error LNK2001: symbole externe non résolu "public: struct g2o::csparse::CSparse::FactorView __cdecl g2o::csparse::CSparse::factor(void)" (?factor@CSparse@csparse@g2o@@QEAA?AUFactorView@123@XZ)
    3>stella_vslam.lib(graph_optimizer.obj) : error LNK2019: symbole externe non résolu "public: bool __cdecl g2o::csparse::CSparse::amd(struct g2o::csparse::CSparse::SparseView const &,class Eigen::Matrix<int,-1,1,0,-1,1> &)" (?amd@CSparse@csparse@g2o@@QEAA_NAEBUSparseView@123@AEAV?$Matrix@H$0?0$00$0A@$0?0$00@Eigen@@@Z) référencé dans la fonction "protected: void __cdecl g2o::LinearSolverCSparse<class Eigen::Matrix<double,7,7,0,7,7> >::computeSymbolicDecomposition(class g2o::SparseBlockMatrix<class Eigen::Matrix<double,7,7,0,7,7> > const &)" (?computeSymbolicDecomposition@?$LinearSolverCSparse@V?$Matrix@N$06$06$0A@$06$06@Eigen@@@g2o@@IEAAXAEBV?$SparseBlockMatrix@V?$Matrix@N$06$06$0A@$06$06@Eigen@@@2@@Z)
[... cut ...] har_traits@D@std@@V?$allocator@D@2@@std@@@Z)

Looking into the details, it seems there are multiple problems:
- `BaseVertex<D, T>::BaseVertex()` is not `inline` (in `g2o/core/base_vertex.hpp`) -> redefinition
- None of the classes/methods of `g2o/solvers/csparse/csparse_wrapper.h` use `G2O_SOLVER_CSPARSE_API` (defined in `g2o_csparse_api.h`), which is a shame because nothing gets exported

I'm not sure about the rest of the redefinitions (`BaseEdge`, `BaseVertex::pop/push`), still investigating.

RainerKuemmerle, 2025-03-18:
I think for some other reason, on the pymem branch I moved all the code to be inline in the template base classes. See, for example, https://github.com/RainerKuemmerle/g2o/blob/pymem/g2o/core/base_vertex.h
Probably should to the same on master.

Thank you as well for spotting the `csparse_wrapper.h` issue. The same is in cholmod's wrapper. Probably, this went unnoticed because the CI does on master not correctly execute the tests on master branch as well.

TLescoatTFX, 2025-03-19:
Thanks, you're fast ! This indeed fixes all link errors about `g2o::csparse::CSparse`.

For the redefinition of the `BaseVertex` constructor, the fix is really easy: all functions in `base_vertex.hpp` should have a literal `inline` keyword. I'm actually surprised it worked on non-windows platforms.

For virtual functions of templated classes (like `BaseVertex::push`), this is hard to solve and I'm not sure how to do it in your project.
I tried several things:
- adding explicit template specialization in `vertex_se3_expmap.h`, as in `types_seven_dof_expmap.h`, bud sadly it does not fix the link error (though I do not understand why). Doing that for `BaseEdge` (in `base_edge.h`) also yielded link errors *inside* g2o, failing the build:

        #ifdef _MSC_VER
        // explicit instantiation of BaseVertex, if not instantiated causes already
        // defined error in some cases (msvc debug only) see links below
        // https://stackoverflow.com/questions/44960760/msvc-dll-exporting-class-that-inherits-from-template-cause-lnk2005-already-defin
        // https://developercommunity.visualstudio.com/content/problem/224597/linker-failing-because-of-multiple-definitions-of.html
        template class G2O_TYPES_SBA_API BaseVertex<6, SE3Quat>;
        #endif
- add explicit imports in the consumer project (which works but is plain dirty and not stable/portable at all):

      #ifdef _MSC_VER
      template class __declspec(dllimport) g2o::BaseEdge<2, g2o::Vector2>;
      template class __declspec(dllimport) g2o::BaseEdge<3, g2o::Vector3>;
      template class __declspec(dllimport) g2o::BaseVertex<6, g2o::SE3Quat>;
      #endif

I can help on this subject, as I have a windows machine with the right setup.

closed 2025-06-05 by github-actions[bot]