RainerKuemmerle/g2o #877: Missing header in install?

skasperski, 2025-03-19:
I am getting this error when building an application that depends on g2o since yesterdays update:
 ```
/opt/workspace/install/slam/g2o/include/g2o/solvers/cholmod/cholmod_wrapper.h:33:10: fatal error: g2o_cholmod_api.h: No such file or directory
       33 | #include "g2o_cholmod_api.h"
          |          ^~~~~~~~~~~~~~~~~~~
```

RainerKuemmerle, 2025-03-19:
Tried to fix linker issues on MSVC.Probably cause this side effect.You can skip last commit on master to avoid the problem.

chhtz, 2025-03-19:
In `g2o/solvers/cholmod/CMakeLists.txt` you need to add `g2o_cholmod_api.h` to the `PUBLIC_HEADERS`-list (if you don't want this header to be public, please don't include it from other public header files).

closed 2025-03-20 by RainerKuemmerle