RainerKuemmerle/g2o #878: Why only generate modularity libraries (libg2o_core.so, libg2o_types_sim3.so) and not libg2o.so

Leven-Cheung, 2025-03-20:
and what i face :(rtgslam) zgz@a207-15-UD:/mnt/15disk2/zgz/slam/RTG-SLAM/thirdParty/pybind/examples$ python orbslam_mono_tum.py
Traceback (most recent call last):
  File "/mnt/15disk2/zgz/slam/RTG-SLAM/thirdParty/pybind/examples/orbslam_mono_tum.py", line 4, in <module>
    import orbslam2
ImportError: libg2o.so: cannot open shared object file: No such file or directory

RainerKuemmerle, 2025-03-20:
I am not exactly sure what you mean. The repo here builds modular libraries. You seem to require a orbslam2 Python wrapper that probably included somewhere along its dependency chain a custom build libg2o.so.
Originally this one: https://github.com/raulmur/ORB_SLAM2/tree/master/Thirdparty/g2o

Maybe, in case you compiled all, you just miss setting up some LD_LIBRARY_PATH or similar for the wrapper to find the so.

In case, you wonder why the libraries here are modular.
At some point, there have been license implications to at least split the GUI and some solvers (as they depend on GPL components).

Leven-Cheung, 2025-03-20:
es, I mean I found out through build that he created a lot of modules like libg2o, but I saw that later compilations need to install libg2o.so, but there is no such a file compiled.

[image]

Leven-Cheung, 2025-03-21:
Find，thank you so much for your reply. i will try to rebuild it by old version.

closed 2025-03-30 by RainerKuemmerle