RainerKuemmerle/g2o #891: Double free or corruption error when destroying optimiser with elements inserted

bomiquel, 2025-05-08:
Hello!

First of all, thank you very much for this library!

I'm relatively new to g2o, so I may be doing something wrong. I'm currently working on **Ubuntu 20.04** and, as a ROS user, I installed the g2o library via:

```bash
 sudo apt install ros-noetic-libg2o
```

## The problem

If I insert elements (vertices or edges) into a `g2o::optimizer` and then allow the optimizer to go out of scope (for example, at the end of a function), the program crashes with the following error:

```
 double free or corruption (out)
 Aborted (core dumped)
```

However, this does **not** happen if the optimizer is empty.

I attempted to use `optimizer.clear()` before destruction, but when I clear the optimizer I get the same error.

## Examples

Here is a simplified example to reproduce the problem when the optimiser goes out of scope:

```cpp
#include <g2o/core/sparse_optimizer.h>
#include <g2o/core/block_solver.h>
#include <g2o/types/slam3d/edge_se3.h>
#include <g2o/solvers/cholmod/linear_solver_cholmod.h>
#include <g2o/core/optimization_algorithm_levenberg.h>

void justInitOptimiser()
{
    std::cout << "Hello justInitOptimiser" << std::endl;

    // Initialise optimiser.
    g2o::SparseOptimizer optimiser;
    std::unique_ptr<g2o::BlockSolverX::LinearSolverType> linear_solver_ptr = std::make_unique<g2o::LinearSolverCholmod<g2o::BlockSolverX::PoseMatrixType>>();
    std::unique_ptr<g2o::BlockSolverX> solver_ptr = std::make_unique<g2o::BlockSolverX>(std::move(linear_solver_ptr));
    std::unique_ptr<g2o::OptimizationAlgorithmLevenberg> solver = std::make_unique<g2o::OptimizationAlgorithmLevenberg>(std::move(solver_ptr));
    optimiser.setAlgorithm(solver.release());

    std::cout << "Bye justInitOptimiser" << std::endl;
}

void insertElement()
{
    std::cout << "Hello insertElement" << std::endl;

    // Initialise optimiser.
    g2o::SparseOptimizer optimiser;
    std::unique_ptr<g2o::BlockSolverX::LinearSolverType> linear_solver_ptr = std::make_unique<g2o::LinearSolverCholmod<g2o::BlockSolverX::PoseMatrixType>>();
    std::unique_ptr<g2o::BlockSolverX> solver_ptr = std::make_unique<g2o::BlockSolverX>(std::move(linear_solver_ptr));
    std::unique_ptr<g2o::OptimizationAlgorithmLevenberg> solver = std::make_unique<g2o::OptimizationAlgorithmLevenberg>(std::move(solver_ptr));
    optimiser.setAlgorithm(solver.release());

    // Dummy first pose.
    Eigen::Quaterniond q(1, 0, 0, 0); // w, x, y, z
    Eigen::Vector3d t(0, 0, 0);
    Eigen::Isometry3d transform = Eigen::Isometry3d::Identity();
    transform.rotate(q);
    transform.pretranslate(t);
    std::cout << "First transformation matrix:\n" << transform.matrix() << std::endl;

    // Add first vertex.
    g2o::VertexSE3* vertex_0 = new g2o::VertexSE3();
    vertex_0->setId(0);
    vertex_0->setEstimate(transform);
    optimiser.addVertex(vertex_0);

    // Dummy second pose.
    t.x() = 1; t.y() = 1; t.z() = 1;
    transform.pretranslate(t);
    std::cout << "Second transformation matrix:\n" << transform.matrix() << std::endl;

    // Add second vertex.
    g2o::VertexSE3* vertex_1 = new g2o::VertexSE3();
    vertex_1->setId(1);
    vertex_1->setEstimate(transform);
    optimiser.addVertex(vertex_1);

    // Add edge.
    g2o::EdgeSE3* edge = new g2o::EdgeSE3();
    edge->setVertex(0, vertex_0);
    edge->setVertex(1, vertex_1);
    edge->setMeasurement(transform);

    std::cout << "Bye insertElement" << std::endl;
}

int main()
{
    std::cout << "Call justInitOptimiser" << std::endl;
    justInitOptimiser();
    std::cout << "Return from justInitOptimiser" << std::endl;

    std::cout << "Call insertElement" << std::endl;
    insertElement();
    std::cout << "Return from insertElement" << std::endl;

    return 0;
}
```

The output:
```
$ ./out_scope_optimiser 
Call justInitOptimiser
Hello justInitOptimiser
Bye justInitOptimiser
Return from justInitOptimiser
Call insertElement
Hello insertElement
First transformation matrix:
1 0 0 0
0 1 0 0
0 0 1 0
0 0 0 1
Second transformation matrix:
1 0 0 1
0 1 0 1
0 0 1 1
0 0 0 1
Bye insertElement
double free or corruption (out)
Aborted (core dumped)
```

Here is a simplified example to reproduce the problem when the optimiser is cleaned:

```cpp
#include <g2o/core/sparse_optimizer.h>
#include <g2o/core/block_solver.h>
#include <g2o/types/slam3d/edge_se3.h>
#include <g2o/solvers/cholmod/linear_solver_cholmod.h>
#include <g2o/core/optimization_algorithm_levenberg.h>

int main()
{
    // Initialise optimiser.
    g2o::SparseOptimizer optimiser;
    std::unique_ptr<g2o::BlockSolverX::LinearSolverType> linear_solver_ptr = std::make_unique<g2o::LinearSolverCholmod<g2o::BlockSolverX::PoseMatrixType>>();
    std::unique_ptr<g2o::BlockSolverX> solver_ptr = std::make_unique<g2o::BlockSolverX>(std::move(linear_solver_ptr));
    std::unique_ptr<g2o::OptimizationAlgorithmLevenberg> solver = std::make_unique<g2o::OptimizationAlgorithmLevenberg>(std::move(solver_ptr));
    optimiser.setAlgorithm(solver.release());

    // Try to clean.
    std::cout << "Before first clear" << std::endl;
    optimiser.clear();
    std::cout << "After first clear" << std::endl;

    // Dummy first pose.
    Eigen::Quaterniond q(1, 0, 0, 0); // w, x, y, z
    Eigen::Vector3d t(0, 0, 0);
    Eigen::Isometry3d transform = Eigen::Isometry3d::Identity();
    transform.rotate(q);
    transform.pretranslate(t);
    std::cout << "First transformation matrix:\n" << transform.matrix() << std::endl;

    // Add first vertex.
    g2o::VertexSE3* vertex_0 = new g2o::VertexSE3();
    vertex_0->setId(0);
    vertex_0->setEstimate(transform);
    optimiser.addVertex(vertex_0);

    // Dummy second pose.
    t.x() = 1; t.y() = 1; t.z() = 1;
    transform.pretranslate(t);
    std::cout << "Second transformation matrix:\n" << transform.matrix() << std::endl;

    // Add second vertex.
    g2o::VertexSE3* vertex_1 = new g2o::VertexSE3();
    vertex_1->setId(1);
    vertex_1->setEstimate(transform);
    optimiser.addVertex(vertex_1);

    // Add edge.
    g2o::EdgeSE3* edge = new g2o::EdgeSE3();
    edge->setVertex(0, vertex_0);
    edge->setVertex(1, vertex_1);
    edge->setMeasurement(transform);

    // Try to clean.
    std::cout << "Before second clear" << std::endl;
    optimiser.clear();
    std::cout << "After second clear" << std::endl;

    return 0;
}
```

The output:
```
$ ./clear_optimiser 
Before first clear
After first clear
First transformation matrix:
1 0 0 0
0 1 0 0
0 0 1 0
0 0 0 1
Second transformation matrix:
1 0 0 1
0 1 0 1
0 0 1 1
0 0 0 1
Before second clear
double free or corruption (out)
Aborted (core dumped)
```

## Question

Am I managing memory incorrectly here? Or is this a bug?

Thanks again for the library and any help you can provide!

RainerKuemmerle, 2025-05-08:
It could be that there is something breaking as you are not actually adding the edge to the optimizer. The g2o code in ROS1 is quite old and did not get updated for a long period. But it can be used in working examples.
Can you please try what happens if you also add the edge to the optimizer.

I tried against master setting up a unit test based on your code. This seems to pass.

bomiquel, 2025-05-15:
Hello again, @RainerKuemmerle!

As you mentioned that the ROS1 version of the g2o code was outdated, I uninstalled it and compiled the latest source code from the master branch of the official g2o GitHub repository in an isolated directory. Along with the library itself, I also compiled the examples and unit tests. 

I ran the `unittest_general` binary, and it completed without any issues. I also tested the `ba_demo` example, and it worked as expected.

However, when I install the library system-wide and link it to my own project, I still encounter the same problems. I even copied `ba_demo.cpp` into my test project and, although it runs correctly, the same error appears at the end of execution.

Here’s the output of `ba_demo` when run directly from the g2o library build directory:
```
~/libs/g2o/build/bin(master)$ ./ba_demo 1 0 1 1 0
PIXEL_NOISE: 1
OUTLIER_RATIO: 0
ROBUST_KERNEL: 1
STRUCTURE_ONLY: 1
DENSE: 0

Performing structure-only BA:

Performing full BA:
iteration= 0	 chi2= 7022.114456	 time= 0.00561613	 cumTime= 0.00561613	 edges= 4911	 schur= 1	 lambda= 1055.091057	 levenbergIter= 1
iteration= 1	 chi2= 7017.961276	 time= 0.00236104	 cumTime= 0.00797717	 edges= 4911	 schur= 1	 lambda= 351.697019	 levenbergIter= 1
iteration= 2	 chi2= 7017.044642	 time= 0.00253313	 cumTime= 0.0105103	 edges= 4911	 schur= 1	 lambda= 117.232340	 levenbergIter= 1
iteration= 3	 chi2= 7016.603161	 time= 0.00235852	 cumTime= 0.0128688	 edges= 4911	 schur= 1	 lambda= 39.077447	 levenbergIter= 1
iteration= 4	 chi2= 7016.154787	 time= 0.00228177	 cumTime= 0.0151506	 edges= 4911	 schur= 1	 lambda= 13.025816	 levenbergIter= 1
iteration= 5	 chi2= 7015.567224	 time= 0.00256659	 cumTime= 0.0177172	 edges= 4911	 schur= 1	 lambda= 4.341939	 levenbergIter= 1
iteration= 6	 chi2= 7014.846392	 time= 0.00249143	 cumTime= 0.0202086	 edges= 4911	 schur= 1	 lambda= 1.447313	 levenbergIter= 1
iteration= 7	 chi2= 7014.372453	 time= 0.00232011	 cumTime= 0.0225287	 edges= 4911	 schur= 1	 lambda= 0.482438	 levenbergIter= 1
iteration= 8	 chi2= 7014.237184	 time= 0.00225565	 cumTime= 0.0247844	 edges= 4911	 schur= 1	 lambda= 0.160813	 levenbergIter= 1
iteration= 9	 chi2= 7014.212244	 time= 0.00237207	 cumTime= 0.0271564	 edges= 4911	 schur= 1	 lambda= 0.053604	 levenbergIter= 1

Point error before optimisation (inliers only): 1.74814
Point error after optimisation (inliers only): 0.0744567
```

And here’s the output when running the same code from within my project:
```
~/catkin_ws/devel/lib/g2o_playground$ ./ba_demo 1 0 1 1 0
PIXEL_NOISE: 1
OUTLIER_RATIO: 0
ROBUST_KERNEL: 1
STRUCTURE_ONLY: 1
DENSE: 0

Performing structure-only BA:

Performing full BA:
iteration= 0	 chi2= 7022.114456	 time= 0.00433003	 cumTime= 0.00433003	 edges= 4911	 schur= 1	 lambda= 1055.091057	 levenbergIter= 1
iteration= 1	 chi2= 7017.961276	 time= 0.00193908	 cumTime= 0.00626911	 edges= 4911	 schur= 1	 lambda= 351.697019	 levenbergIter= 1
iteration= 2	 chi2= 7017.044642	 time= 0.00196052	 cumTime= 0.00822964	 edges= 4911	 schur= 1	 lambda= 117.232340	 levenbergIter= 1
iteration= 3	 chi2= 7016.603161	 time= 0.00198943	 cumTime= 0.0102191	 edges= 4911	 schur= 1	 lambda= 39.077447	 levenbergIter= 1
iteration= 4	 chi2= 7016.154787	 time= 0.00200592	 cumTime= 0.012225	 edges= 4911	 schur= 1	 lambda= 13.025816	 levenbergIter= 1
iteration= 5	 chi2= 7015.567224	 time= 0.001971	 cumTime= 0.014196	 edges= 4911	 schur= 1	 lambda= 4.341939	 levenbergIter= 1
iteration= 6	 chi2= 7014.846392	 time= 0.00195347	 cumTime= 0.0161495	 edges= 4911	 schur= 1	 lambda= 1.447313	 levenbergIter= 1
iteration= 7	 chi2= 7014.372453	 time= 0.00196653	 cumTime= 0.018116	 edges= 4911	 schur= 1	 lambda= 0.482438	 levenbergIter= 1
iteration= 8	 chi2= 7014.237184	 time= 0.00192581	 cumTime= 0.0200418	 edges= 4911	 schur= 1	 lambda= 0.160813	 levenbergIter= 1
iteration= 9	 chi2= 7014.212244	 time= 0.00198196	 cumTime= 0.0220238	 edges= 4911	 schur= 1	 lambda= 0.053604	 levenbergIter= 1

Point error before optimisation (inliers only): 1.74814
Point error after optimisation (inliers only): 0.0744567

double free or corruption (out)
Aborted (core dumped)
```

Is it possible that even if it doesn't give any compilation error, the error is in the `CMakeLists.txt`? (My project is catkinised in order to be able to use ROS later.) Here's the relevant part of the `CMakeLists.txt`:
```
cmake_minimum_required(VERSION 3.0.2)
set(CMAKE_MODULE_PATH ${CMAKE_MODULE_PATH} "${CMAKE_CURRENT_SOURCE_DIR}/cmake-modules/")
project(g2o_playground)
[... 27 lines cut ...]
)

include_directories(
    include
    ${catkin_INCLUDE_DIRS}
    ${SUITESPARSE_INCLUDE_DIRS}
    ${G2O_INCLUDE_DIRS}
    ${EIGEN3_INCLUDE_DIRS}
)

add_executable(ba_demo src/ba_demo.cpp)
add_executable(clear_optimiser src/clear_optimiser.cpp)
add_executable(out_scope_optimiser src/out_scope_optimiser.cpp)

add_dependencies(ba_demo ${catkin_EXPORTED_TARGETS})
add_dependencies(clear_optimiser ${catkin_EXPORTED_TARGETS})
add_dependencies(out_scope_optimiser ${catkin_EXPORTED_TARGETS})

target_link_libraries(ba_demo ${catkin_LIBRARIES} ${EIGEN3_LIBRARIES} ${SUITESPARSE_LIBRARIES} ${G2O_LIBRARIES} cholmod)
target_link_libraries(clear_optimiser ${catkin_LIBRARIES} ${EIGEN3_LIBRARIES} ${SUITESPARSE_LIBRARIES} ${G2O_LIBRARIES} cholmod)
target_link_libraries(out_scope_optimiser ${catkin_LIBRARIES} ${EIGEN3_LIBRARIES} ${SUITESPARSE_LIBRARIES} ${G2O_LIBRARIES} cholmod)
```
Is it possible that, even though everything compiles fine, the issue is caused by something in the way the library is linked or used in the catkin environment?

Thank you again for your time and help!

RainerKuemmerle, 2025-05-15:
march=native ist probably the issue.This needs to be the same flag on all compilation units. This has been an issue in the past as well.I assume that the flag changes internals of the eigen library causing undefined behavior if one header is translated with native and some others are not.

bomiquel, 2025-05-19:
That was the problem! I removed `-march=native` from both the dummy project and the main project, and now they are both running perfectly. Thank you very much, @RainerKuemmerle!

closed 2025-05-19 by bomiquel